Ultimaker/Cura #15514: Slicing Error

DustyNo99, 2023-05-13:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

1. I hit slice for a model of a dice tower door, but it wouldn't complete slice.
2. I tried again but got an error message with a link to report the issue.
3. Tried several more times with different parameters, but still got the error message.

### Actual results

I would receive an error message, and the model wouldn't slice.

### Expected results

The model should have been sliced and ready to print.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Cura Error on Slice.zip](https://github.com/Ultimaker/Cura/files/11469015/Cura.Error.on.Slice.zip)
[image]
[image]
[cura.log - Copy.zip](https://github.com/Ultimaker/Cura/files/11469040/cura.log.-.Copy.zip)

GregValiant, 2023-05-13:
Thanks for the report and the files.
I'm not sure what to make of this one.  MS 3D Builder indicated the model has errors as does formware.co.  PrusaSlicer indicated there are 497316 open edge errors in the model and then it didn't have any trouble slicing it.  In Cura when I switched from your Elegoo to my Ender the model sliced.
Back to your Elegoo I made a couple of changes to the setup (one was to turn off supports) and the model sliced.

It might be the model or it might be your settings.  The Cura team will take a look.

madhephaestus, 2023-10-03:
For reference i sliced this model in 5.4.0 and it sliced correctly

GregValiant, 2024-06-22:
I'll go ahead and close this today.

closed 2024-06-22 by GregValiant