Ultimaker/Cura #15515: Print Speed incorrect when sliced compared to what was set

LazZer-, 2023-05-13:
### Application Version

5.3.1

### Platform

Win10, nVidia GTX1660 Super

### Printer

Custom (Craftbot Plus Classic)

### Reproduction steps

1. Plopped model into Cura
2. Used the slicing profile i've been for ages, migrated from Cura 4.x.x at least with the print speed set to 75mm/s (except for travel moves)

### Actual results

The inner wall speed tops out at 93.75mm/s according to the preview, and on the printer too.

### Expected results

The max print speed being 75mm/s, as set.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[CFFFP_pirosszuper.zip](https://github.com/Ultimaker/Cura/files/11469367/CFFFP_pirosszuper.zip)

GregValiant, 2023-05-13:
Thanks for the report and the file.
I see that the speed is bouncing around a lot.  In the experimental section you can set the "Flow Rate Compensation Factor" to "0" and the speeds will be more consistent (but still not exact).

The Skins and Outer Walls are at your set speeds.  It is the Inner-Walls that vary and where that "93" speed is.  The left column is the line number within the gcode file and the fifth column is Feed Rate in mm/sec.  This is with "Flow Compensation" set to 0.
Line #||	E Value||	Delta E||	Layer||	Feed Rate||	Vol mm³||	Act Extr Length||	Time||	FlowRate mm³/sec||	LW from Vol/LH||	Type

84414      0.02990      0.02990      	;LAYER:23        83     0.07192     0.54597     0.00658     10.93315     0.44     WALL-INNER
84415	0.01731		0.01731	;LAYER:23	93	0.04164	0.35396	0.00381	10.93947	0.39	WALL-INNER
84416	0.02698		0.02698	;LAYER:23	82	0.06489	0.48411	0.00590	10.99204	0.45	WALL-INNER
84417	0.04391		0.04391	;LAYER:23	77	0.10562	0.74230	0.00964	10.95572	0.47	WALL-INNER
84418	0.03369		0.03369	;LAYER:23	87	0.08103	0.64399	0.00740	10.94733	0.42	WALL-INNER
84419	0.02962		0.02962	;LAYER:23	83	0.07124	0.54083	0.00652	10.93368	0.44	WALL-INNER
The extrusion length of that 93mm/sec extrusion is 0.35396mm and it has the smallest line width in that group at 0.39.  Taking Accel and Jerk into account the average speed from start to end is much lower than 93.  My guess would be around 75.

Most of the Inner Walls are like this:
84571	0.05647		0.05647	;LAYER:23	74	0.13583	0.91300	0.01234	11.00897	0.50	WALL-INNER
84572	0.03263		0.03263	;LAYER:23	75	0.07848	0.54033	0.00720	10.89401	0.48	WALL-INNER
84573	0.06203		0.06203	;LAYER:23	76	0.14920	1.03353	0.01360	10.97134	0.48	WALL-INNER
84574	0.06890		0.06890	;LAYER:23	75	0.16572	1.12742	0.01503	11.02459	0.49	WALL-INNER
84575	0.03641		0.03641	;LAYER:23	75	0.08758	0.59579	0.00794	11.02448	0.49	WALL-INNER
84576	0.04817		0.04817	;LAYER:23	74	0.11586	0.78653	0.01063	10.90086	0.49	WALL-INNER
84577	0.05570		0.05570	;LAYER:23	74	0.13397	0.90230	0.01219	10.98756	0.49	WALL-INNER
84578	0.06881		0.06881	;LAYER:23	75	0.16551	1.12584	0.01501	11.02559	0.49	WALL-INNER

It is the Inner Walls that Cura adjusts the line width for.  There must be something within the code that is adjusting the speed.  Whether it is simply Accel and Jerk or some other factor, I don't know.
The Cura team will take a look.

GregValiant, 2024-06-22:
I'll close this today.

closed 2024-06-22 by GregValiant