Ultimaker/Cura #15516: User Specific Z-Seam position faulty. Especially on multiple similar objects.

NVNDO, 2023-05-13:
### Application Version

5.2.2

### Platform

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Slice a model and multiply it (or directly import more of the same objects)
2. Set Z Seam Alignment to - User Specific
3. Set Z Seam Position to - Whatever. 

### Actual results

The Seams is NEVER placed at the same spot regardless of which position is set.  
[image]
[image]

### Expected results

All Seams being the same and located at set location. 
In my case = All Seams should have been at the "back" 
(and if possible "straight" & all shaped the same way - when objects are the same)

Prusa Slicer is perfect in this regard.
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Z Seam issue.zip](https://github.com/Ultimaker/Cura/files/11469558/Z.Seam.issue.zip)

smartavionics, 2023-05-13:
You need to set Z Seam Relative.

GregValiant, 2023-05-13:
Thanks for the report and the file.
A problem is that you don't have all the settings visible.  Use the button to the right of the Settings Search box and set the visibility to "All".
Notice that when set to "Back" that the "Z Seam X" is mid-point of the back of the printer.  All your seams are pointing to that location.
Enable "Z seam Relative" and you will see that all the seams move to point squarely at the back edge rather than at the mid-point of the back edge.
Your project only has the round objects so I can't check the others.
I'm inclined to remove the bug label from this as it appears to be "expected behavior" and a settings issue.  Let me know if you think that is premature or if you have continuing problems after enabled "Z Seam Relative".

This is the way you have the project set up:
[image]

The only change here is "Z Seam Relative"
[image]

NVNDO, 2023-05-13:
Hey @GregValiant,

I sincerely did not know about this Z Seam Relative function. It seems to work, so THX a lot already!!  ...though not as perfect as Prusa Slicer as the Seam still varies a little here and there. I cant understand why, I might just be good enough compared to previous, but it still bothers me a little that they aren't the same on all objects.

[image]

[image]

The Clips though are spot on!

[image]

Playing around, I noticed that the "Z Seam X" and "Z Seam Y" function conflicts with the positioning of Z Seam, so it seems to be if someone wants to specifically move it to a certain angle, right?

GregValiant, 2023-05-13:
Not a certain angle, but a certain location in regards to the entered numbers.  
When "Z Seam Relative" is enabled the numbers relate to the mid-point of each part.
When "Z Seam Relative" is off then the numbers relate to the build plate.
So it can appear to be an angular change, but it is really a reaction to whether the point in space is in regards to the part (relative), or the printer (absolute).

NVNDO, 2023-05-13:
> You need to set Z Seam Relative.

You were totally right! A huge thx also to you!

NVNDO, 2023-05-13:
> Not a certain angle, but a certain location in regards to the entered numbers. When "Z Seam Relative" is enabled the numbers relate to the mid-point of each part. When "Z Seam Relative" is off then the numbers relate to the build plate. So it can appear to be an angular change, but it is really a reaction to whether the point in space is in regards to the part (relative), or the printer (absolute). If the X and Y are linked to the Z seam location setting and if Z Seam Relative is turned off then "Left Front" would be shown as X0 Y0. But if you turn on Z Seam Relative you will see that the location changes to X-115 Y-115 (which are the mid-point coordinates of my printer).

Now I do fully understand it, thx to your very in-depth explanation, thank you so much! 

In regards of my pickiness, I assume its like it is. Probably good enough for 99% of the population anyways, but if in any case you or someone of the team is bored, perfectioning the outcome so all Z Seams are the same, would be awesome.

GregValiant, 2023-05-13:
There are a couple of other bug reports about the Z seam "wandering" so the Cura Team is aware of the problem.  They have a lot on their plate with the slicing errors so I have no idea when they will be able to get to this.

NVNDO, 2023-05-13:
I can imagine....

My main issue here is, that the part im printing is slit in the middle. Once I print and reconnect them, I want to have the seam lining up. but I just realized that this is not going to happen, neither with Z Seam Relative as its not accurate enough and doe snot place the Seam in the exact "back" or 117.5 (middle of my print-bed)

A shame something so basic is so hard to accomplish..