Ultimaker/Cura #15518: Slicing Open Gaps Closed

secondkings2-24, 2023-05-14:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

Ender V2

### Reproduction steps

Import close pin that can open. There is a gap so it can do this. It appears open when previewed. 
Press "slice". After slicing there is no gap. 

https://www.thingiverse.com/thing:1766868/files

### Actual results

Press "slice". After slicing there is no gap. The slicer remo0ves the gap completely

### Expected results

There is supposed to be a gap so that it can open.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[CE3E3V2_Clothespin_UPDATED.zip](https://github.com/Ultimaker/Cura/files/11470999/CE3E3V2_Clothespin_UPDATED.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11471000/cura.log)

GregValiant, 2023-05-14:
Thanks for the report and the files.
I can duplicate this with your project.  Setting the "Minimum Line Width" to .25 keeps Cura within the lines.  I don't think that should be necessary.  The gap on the "thicker spring" model is 0.25mm and may just be too small to keep Cura from becoming confused.  I think it should still respect the walls though.  If there was some inadvertent smearing that connected across the gap that would be one thing.  Acting like the gap isn't there is a whole different thing.
The Cura team will take a look.  This behavior is similar to [THIS POST](https://community.ultimaker.com/topic/43537-cura-ignoring-small-gap/#comment-322350) on the UM forum that describes the same sort of problem.
This is with the minimum line width adjusted to 0.259 (at .260 I get the same effect that you did).
[image]

secondkings2-24, 2023-05-15:
Build 5.3.0 does it too. I also had a different problem with supports and
radius print. Do you want a separate bug report for those too?

Aaron

"He is no fool who gives what he cannot keep to gain that which he cannot
lose."
- Jim Elliot

GregValiant, 2023-05-15:
A project file would be good.  The more examples the greater the chance of chasing this down.

barkermn01, 2023-12-28:
[Logitech_C270_Mount_for_Ender_3_v2.zip](https://github.com/Ultimaker/Cura/files/13789976/Logitech_C270_Mount_for_Ender_3_v2.zip)

As your asking for models, this is one that does it strangly the left hand edge
[image]

GregValiant, 2023-12-29:
@barkermn01 For that particular model the problem went away when I turned on "Print Thin Walls".  Unfortunately, "Print Thin Walls" has been linked to other bugs in Cura.  In this case the slice looked good.
[image]

I went back and loaded the other two models in this report and they also sliced correctly with "Print Thin Walls" turned on.
[image]