Ultimaker/Cura #15519: Combing travel moves are extremely low resolution, leading to blobs

daniels220, 2023-05-14:
### Application Version

5.3.1

### Platform

MacOS

### Printer

Creality CR-10s

### Reproduction steps

Have a model with thin, curved sections (thin enough to consist of all wall lines—things get even weirder if only outer walls are present, but this issue is about parts that are e.g. 4-6 lines wide), and wider areas on either side such that the printer must travel across the narrow section.

### Actual results

The combing travel move is broken up into only a few segments, so it does not follow the curvature of the model, so much so that the nozzle exits and re-enters the model and leaves blob(s) on the surface—exactly what combing is trying to avoid! This is not affected by the "maximum travel resolution", nor the travel avoid distance (though I wouldn't really expect this to matter as we're not "avoiding"—I mention it for completeness' sake).

[image]

### Expected results

The travel move should exactly follow the curvature of the walls, ideally to the inside of at least the first inner wall if there is one, or some multiple of the nozzle width from the edge, etc. I _would_ expect this to be subject to the maximum travel resolution, but that should only come into play for very large values of that setting.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I am using Cura 5.3 but have noticed issues like this for a long time—not sure if only since Cura 5 or all along.

[Snap clip.3mf.zip](https://github.com/Ultimaker/Cura/files/11471023/Snap.clip.3mf.zip)

GregValiant, 2023-05-14:
Thanks for the report and the file.
I think "Avoid printed parts when traveling" is causing a conflict with your combing mode.  On the one hand Cura is trying to move outside the part in order to Avoid, and on the other it's trying to stay "Not in Skin" of which there really isn't any as this model consists of only walls.  It would seem that adjusting "Travel Resolution" has no effect.

I noticed that switching to either a .4 or a .6 nozzle fixes the problem.  It could be a setting in a quality file that is in the background that is effecting the .5 nozzle.  That could make this a Creality issue (or whoever submitted the definition files).  Simply changing the Line Width to either .4 or .6 had no effect...I had to change the nozzle size.
I'll leave the bug label on this and the Cura team will take a look.

This is with a .4 nozzle and "Avoid printed parts..." is turned off.  For this particular model the print time remains the same as with the .5 nozzle.
[image]

daniels220, 2023-05-15:
In short—I discovered the same thing you did about the nozzle size, and in fact I think that's the only real factor here. For me, changing combing mode to "All" and/or turning off "Avoid printed parts when traveling" has no effect on the slice. And as you said, a 0.6 or 0.4 nozzle does not show this problem while 0.5 does.

I don't think it's so much a difference in the 0.5 configuration file as a the-numbers-are-juuuust-wrong issue—that section of the part is nominally 2.0mm thick, but being a mesh the "actual" width flutters around that value. I think this being a multiple of the nozzle size...well, hmm, it's a multiple of 0.4mm as well—maybe an _even_ multiple? Or the fact that the line width is also 0.5? Anyway, something about that is the cause of the problem. Part of my suspicion comes from the difference I see between 0.6 and 0.4—0.6 follows the inside of the outer wall, while 0.4 does the IMO correct thing and follows the middle of the section. (IMO combing should always travel inside at least one inner wall when possible.) 0.6:

[image]

0.4, as it is for you:

[image]

----

This isn't the first time I've encountered issues that are sensitive to the nozzle size even when the other obvious settings that depend on it (line width and minimum line widths, mainly) are held constant. Is there a quick way to find out what exact other settings are affected by the nozzle size? A sort of reverse-dependencies lookup. Also, are there any places in the slicer code that refer directly to the nozzle size, and thus depend on it no matter what the other settings values are?