Ultimaker/Cura #15520: no internal support for roof corners

doctoryo12345, 2023-05-14:
### Application Version

5.2.1

### Platform

windows 10

### Printer

creality ender 3 s1

### Reproduction steps

1.placed a model inside cura. The model was 2 rectangular surfaces on top of each other, each rectangle is around 5mm in z height; base was wider than the top rectangle.
2. placed petg settings with 3% infill.
3. upon slicing, you would see that the second rectangle angles will print on top of infills. some layers will not have support from the build plate

### Actual results

it didn't print properly

### Expected results

internal support should be placed for all roof edges or slicer should put a pseudo roof.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

[image]

GregValiant, 2023-05-14:
Thanks for the report.
It isn't really clear what you are looking for/at.  Use the "File | Save Project" command to create a 3mf project file.  Zip the file and post the zip folder here.

Asterchades, 2023-05-15:
I believe they're referring to the inner corners of the fresh top layer which occur over gaps in the infill, as seen towards the lower left and right of the model preview.

This is going to happen when you use a low infill percentage, regardless of slicer. The same will happen with supports at a low density outside of the model, with the only variation being that supports have the option for an interface (the top layers _are_ the interface for infill). Infill (as with support) is always generated as a "dumb" structure - it generates with consistent placement across an entre model, which means it doesn't know nor care if it's under an extended flat surface or a corner of any type.

You could try Gradual Infill Steps or Skin Edge Support to help it out, but realistically the solution is a higher infill density so that the corners are more likely to be over existing material.

closed 2023-05-29 by github-actions[bot]