Ultimaker/Cura #15523: slicing in ultimaker

Hybred1689, 2023-05-14:
### Application Version

5.3.1

### Platform

windows 11

### Printer

Ultimaker Cura

### Reproduction steps

imported a file from Armorsmith and tried to slice the file 

### Actual results

it says slicing failed with an unexpected error

### Expected results

completes slicing to print file

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3E3V2_IstelleMHELMET_V5_FULL0.zip](https://github.com/Ultimaker/Cura/files/11472908/CE3E3V2_IstelleMHELMET_V5_FULL0.zip)

GregValiant, 2023-05-15:
Thanks for the report and the file.
It wouldn't slice for me either.  I exported the model and checked for errors.  This is the view in Mesh Mixer.  Each ball represents an error in the mesh.
[image]

[This repair site](https://formware.co/OnlineStlRepair) provided this report.
-> Analysed your file:
--> 12 Naked edges (?)
--> 3 Planar holes (?)
--> 0 Non-planar holes (?)
--> 306 Non-manifold edges (?)
--> 324 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 206 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 28514ms ------
-> Vertex count changed from 71844 to 75823 (+3979)
-> Triangle count changed from 144022 to 153026 (+9004)

After all that I still had to move and rotate the repaired model to get it to slice.
The Cura team will take a look.
[image]

Hybred1689, 2023-05-16:
Thank you so much for a timely response! I appreciate the effort

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic