Ultimaker/Cura #15524: Cura crashes on startup "Type UM.SimpleButton unavailable"

TahaAbbasi-Hashemi, 2023-05-15:
### Application Version

cura version: 5.2.2

### Platform

Alpine Linux

### Printer

Not applicable

### Reproduction steps

1. Ran the command "cura" in the terminal. At which point the program crashes

### Actual results

The error message comes in the terminal

"2023-05-15 00:41:27,784 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [616]: file:///usr/lib/cura/plugins/ModelChecker/ModelChecker.qml:8:1: Type UM.SimpleButton unavailable
2023-05-15 00:41:27,785 - ERROR - [MainThread] UM.Qt.QtApplication.createQmlComponent [616]: file:///usr/lib/python3.11/site-packages/UM/Qt/qml/UM/SimpleButton.qml:30:5: ColorImage is not a type"

### Expected results

Cura should open

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11473324/cura.log)

vanitasvitae, 2023-05-15:
I'm having the same issue on Fedora 38 with Cura 5.3.0.

nallath, 2023-05-15:
Sounds a lot like a packaging issue. I've seen those errors before on my local dev env from source.

MariMakes, 2023-05-19:
Hey @TahaAbbasi-Hashemi,

Thanks for your report 👍
I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

rosbergrodrigues, 2023-05-20:
I'm using Fedora 38, and having same problem.
Waiting for solution. too ;)

fieldOfView, 2023-05-20:
That `ColorImage` on line 30 of `.../site-packages/UM/Qt/qml/UM/SimpleButton.qml` should probably be `UM.ColorImage`. And there should be an `import UM 1.5 as UM` statement near the top of the file. 

Strange that it crashes for you (as it should) but not for the majority of users.

rosbergrodrigues, 2023-05-20:
> That `ColorImage` on line 30 of `.../site-packages/UM/Qt/qml/UM/SimpleButton.qml` should probably be `UM.ColorImage`. And there should be an `import UM 1.5 as UM` statement near the top of the file.
> 
> Strange that it crashes for you (as it should) but not for the majority of users.

Yeah!
I edited the file:

sudo nano -c /usr/lib/python3.11/site-packages/UM/Qt/qml/UM/SimpleButton.qml

1- add "import UM 1.5 as UM"
2 - add "UM." on ColorImage

// Copyright (c) 2018 Ultimaker B.V.
// Uranium is released under the terms of the LGPLv3 or higher.

import QtQuick 2.1
import UM 1.5 as UM

MouseArea
{
    id: base

    hoverEnabled: true

    property color color: "black"
    property color hoverColor: color
    property color backgroundColor: "transparent"
    property color hoverBackgroundColor: backgroundColor
    property alias iconSource: image.source
    property real iconMargin: 0

    property alias hovered: base.containsMouse
    property alias backgroundRadius: background.radius

    Rectangle
    {
     	id: background
        anchors.fill: parent
        color: base.containsMouse ? base.hoverBackgroundColor : base.backgroundColor
        radius: 0
    }

    UM.ColorImage
    {
     	id: image
        anchors.fill: parent
        anchors.margins: base.iconMargin
        color: base.containsMouse ? base.hoverColor : base.color
        visible: source != ""
    }
}

And it Works!! Thank you so much

One more doubt, would i, add UM. on Rectangle line too??

fieldOfView, 2023-05-20:
Thanks for testing. I will make a pull request so this fix can be incorporated in the next release.

`Rectangle` is a standard QML element, or actually part of the QtQuick package. `ColorImage` is defined by the [Uranium](https://github.com/ultimaker/uranium) package, which is imported as `UM`. That's why `ColorImage` should be prefixed with the `UM` class.

fieldOfView, 2023-05-20:
See https://github.com/Ultimaker/Uranium/pull/874