Ultimaker/Cura #15531: Slice errors out when slicing autoscale.

DuckersMcQuack, 2023-05-15:
### Application Version

5.3.0

### Platform

windows 11

### Printer

Rook MK1, so pretty much custom

### Reproduction steps

Simply load in crystal dragon from cults, and press slice. Bed is 102x111, so figure was sized down automatically.

### Actual results

As image shows, and when i size it down to 40% from 51% auto, it never goes past 50% sliced status.

[image]

### Expected results

Model to successfully slice.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Cura didnæt generate a log from the error. Last log was from days ago.

GregValiant, 2023-05-17:
Thanks for the report.
There are several "Crystal Dragons" out there and more than one has showed up here in a Cura bug report ( #15362 and #12546 ).  Both of the models that were submitted were riddled with errors.  If you would include a project file I can take a look.

DuckersMcQuack, 2023-05-18:
Huh indeed. And apparently microsoft's 3d builder couldn't fix them/find the relevant errors. Know of a program that can properly repair them? Or is this "don't mind the errors" from cura 4 could be ported to 5? As i can slice the dragon twice the size on cura 4.12.1 that i use for my ender just fine. Just latest 2 cura versions of 5 that i use for my rook that struggles.

Here's a stupidly tall 0.08 castle with 0.2 nozzle i tested to slice in 5.3.1, it went through just fine.
[image]

GregValiant, 2023-05-19:
I'm just the greeter here.  I don't have any connection to UltiMaker and I don't know the inner workings of the program.  I have no idea why some models slice fine and others won't.  I used Mesh Mixer to check the Crystal Dragons.  I probably used [formware.co](https://formware.co/OnlineStlRepair) for the repairs (I can't remember) but that site has a 20mb size limit.
Cura has no "auto-repair" function like PrusaSlicer and IdeaMaker so that leaves it up to the user to check their models.
I think one of the Crystal Dragon bug reports was actually filed by the designer.  Guessing from what I saw in the error picture in Mesh Mixer it looked like he unknowingly had one bad section of the dragon but then copied it and used it several times to make the tail.
Speaking of nozzle size and layer height, one of the Cura bugs can be defeated by changing the "Initial Layer Height" by 0.01mm.  It's enough to get the slices away from whatever was causing the glitch.