Ultimaker/Cura #15534: Slicing Failed

TiredoftheBull, 2023-05-16:
### Application Version

5.1.0

### Platform

Windows 11

### Printer

Monoprice MP Select Mini IIIP

### Reproduction steps

Opened stl file.
Rotated model 180 degrees.
Selected "Fine" profile.
Pushed "Slice"

### Actual results

Slice Failed

Slicing failed with an unexpected error.  Please consider reporting a bug on our issue tracker.

### Expected results

Sliced model to export for printing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Submit.zip](https://github.com/Ultimaker/Cura/files/11484148/Submit.zip)

GregValiant, 2023-05-17:
Thanks for the report and the file.
In this screenshot you can see red inside the model.  Cura paints upper surfaces red when the model has errors.
[image]

I uploaded the model to formware.co/onlinestlrepair and here is the report.
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 23561ms ------
-> Vertex count changed from 15364 to 28092 (+12728)
-> Triangle count changed from 30724 to 68874 (+38150)

It didn't find any errors BUT it had to add a LOT of vertices and triangles.
Here is your model on the left and the repaired model on the right.  Your model is watertight but it shouldn't have those red surfaces which Cura is indicating are in error.
[image]

Here is the repaired model sliced.
[image]

Although the model is Watertight, it has issues.  I'll attached the repaired model so you can see if it makes a difference on your computer.  Let us know if it will slice for you.  I don't think this is a Cura bug but rather a model problem.  I'll leave the bug label on this for now.

[Bottom_fixed.zip](https://github.com/Ultimaker/Cura/files/11501693/Bottom_fixed.zip)

TiredoftheBull, 2023-05-18:
Thanks for the help. Unfortunately the Botton_fixed.stl file also failed to slice.

GregValiant, 2023-05-18:
I'll mark this as a duplicate slicing error then.
The Cura team may want to know more about your system.  I see you are using Win11...which version (home or pro?).
What processor and video system does the computer have and how many monitors are you using?

TiredoftheBull, 2023-05-18:
Dell G5 5590 Laptop
Windows 11 Home (X64) Build 22621.1702
Intel Core i7-9750H
NVIDIA GeForce RTX 2060
16GB PC4-213000 DDR4 SK Hynix (2 of HMA81GS6JJR8N-VK)
HP HPN360C
Monoprice MP Select Mini 3D Printer V1

TiredoftheBull, 2023-05-20:
A little extra information.  I installed Cura on my Ubuntu partition and that version of Cura sliced the model fine (older, I'll check the version). The print had a lot of extra material (I called it slag) in the area around the holes which is likely the section causing the newer Cura to fail to slice.