Ultimaker/Cura #15540: Bug in Small Layer Fan Settings for Dual Head Printing

trussell322, 2023-05-16:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Ultimaker 5S

### Reproduction steps

I sliced a dual head "C" shaped part. Thus, there were some small layers and some larger layers to be printed with Extruder 1, and some support to be printed with Extruder 2. I had the cooling settings enabled.

### Actual results

The fan settings for short layer times (small layers) are incorrect every other layer. I have attached a picture to show what I mean. The picture was produced using PrusaSlicer and it's actually a picture of a slice made for an Essentium HSE 280i machine, but I saw similar behavior when I tried slicing for an Ultimaker 5S machine.

[image]

### Expected results

In the picture you can see a subtle gradient in the fan speed % every other layer (the reddish-looking layers which have a slightly darker shade for smaller layers). This is what I'm looking for. But the blue layers in between appear to be incorrectly using the default value of the "Fan Speed" setting, as opposed to being overridden by the small layer time settings.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I'm attaching a project file for both the Essentium HSE 280i and the Ultimaker S5 where this issue can be seen.

[C_Parts.zip](https://github.com/Ultimaker/Cura/files/11492652/C_Parts.zip)

GregValiant, 2023-05-17:
Thanks for the report and the file.
Given your settings what you have is "expected behavior".  Your "Layer Height" is 0.15mm but your "Support Infill Thickness" is 0.3mm so support infill is going down every other layer.  If you check the "Elapsed Time" for each layer you will find:
Layer:71 - 25 seconds
Layer:72 - 08 seconds
Layer:73 - 25 seconds
Layer:74 - 08 seconds
Layer:75 - 25 seconds
Layer:76 - 08 seconds
Layer:77 - 25 seconds
Layer:78 - 08 seconds
So you can see that yes, every other layer is getting Max fan but the longer layers where Support Infill is also being printed don't qualify.

I'll go ahead and remove the bug label as this is a settings issue but I'll leave it open so you can respond.
What I will do to help is pass along my proposed post processor "Advanced Cooling Fan Control".  You can set the fan speed "By Feature" and you should get what you need.
Extract the AdvancedCoolingProfile.py file and put it in "C:\Program Files\UltiMaker Cura 5.3.1\share\cura\plugins\PostProcessingPlugin\scripts" folder.
It will be available under "Extensions/PostProcessing/Modify Gcode" and then "Add a Script" (after you restart Cura).  You can run more than one instance and run "By Layer" up to layer 34, run "By Feature" to layer 134, and then run "By Layer" the rest of the file.  The second and third instances must have "Remove M106 Lines" disabled or it will wipe out the fan commands that the previous instances inserted.
[AddCoolingProfile.zip](https://github.com/Ultimaker/Cura/files/11493165/AddCoolingProfile.zip)

This is in PrusaSlicer.  I've post processed the file with "By Layer" at 50% to layer 33 then with "By Feature" cooling from layer 34 to 134.  You can see that the model is 100% and the support is at 35% on the same layers, and then it finishes at 50%.  It might not be as fashionable as stripes, but it may print better.
[image]

trussell322, 2023-05-17:
Hi Greg, thanks a lot for the feedback. I must have made a mistake in sending the wrong 3mf to you before though, as I have seen this issue even when my layer height is the same for both tool heads. I'm attaching a 3mf and the gcode that goes with it this time too, to make sure you are seeing what I'm seeing. I've made the layer height a constant 0.15 mm and from Layer 71-80 you can see that the layer times are all about 24 seconds:
;LAYER:70
...
;TIME_ELAPSED:3025.819258
;LAYER:71
...
;TIME_ELAPSED:3049.946568
;LAYER:72
...
;TIME_ELAPSED:3074.151530
;LAYER:73
...
;TIME_ELAPSED:3098.261856
;LAYER:74
...
;TIME_ELAPSED:3122.457264
;LAYER:75
...
;TIME_ELAPSED:3146.559989
;LAYER:76
...
;TIME_ELAPSED:3170.747566
;LAYER:77
...
;TIME_ELAPSED:3194.842957
;LAYER:78
...
;TIME_ELAPSED:3219.022997
;LAYER:79
...
;TIME_ELAPSED:3243.110645
;LAYER:80
...
;TIME_ELAPSED:3267.284369

By the way, I've actually gotten your plugin from you before and it's great! However, it seems to be useful only on a case-by-case basis. Ideally, I'm looking for a fix to the small layer settings. This way, the solution will be generalized and I can share my profile with others to use on arbitrary geometries without having to be taught how to use your plugin. The picture you've given looks like what I'm going for - now, how can we get the small layer settings to produce those results for us without having to use your plugin to man-handle the gcode on a part-by-part basis?

Thanks for your help.

[gcode_and_3mf_with_problem.zip](https://github.com/Ultimaker/Cura/files/11501530/gcode_and_3mf_with_problem.zip)

GregValiant, 2023-05-18:
I played with it some more.  
Here is your project
[image]

The stripes occur because there is an M107 line in the gcode _every other layer_ through the neck area.  The fan is on for one layer and off for the next.  That sure looks like a bug to me.

In this shot I've enabled Search and Replace and commented out all the M107 lines.  The stripes are gone.
[image]

I'll put the bug label back on and someone from the Cura team will take a look.

GregValiant, 2023-05-18:
Heh, it only took two of us about 3 hours each to figure it out.

trussell322, 2023-05-18:
Also, I believe the M107 commands were added by Cura since the 'Regular Fan Speed' setting was set to 0. But if you make the Regular Fan Speed > 0, I believe it will wrongly add M106 S__ commands every other layer instead of M107 commands. (I would have expected 'M106 S0' as opposed to 'M107,' but that doesn't seem to be the case.)

trussell322, 2023-05-18:
I also noticed that the layers that have the correct fan speed are after a tool change. But since tool head 1 will print two layers, then tool head 2 will print two layers, etc...the second layer a tool head prints before it does a tool change is the one that seems to have the issue.