Ultimaker/Cura #15541: cura 5.3.1 does not save with gcode extensions

Miahotrod, 2023-05-17:
### Application Version

5.3.1

### Platform

Linux Mint 21 Cinimon 

### Printer

Tronxy X5SA/X5ST/Pro/2E 400mm

### Reproduction steps

1 Open a STL file and Slice it 
2 Save file in GCODE format save to disk 
3 file will save with TXT format not and does not have .GCODE extension
Other notes the Printer does not recognize with out extension and when I add the .GCODE extension the Printer goes outside printer area and jams. 

### Actual results

File saves in TXT format with out .GCODE extension 
and unrecognizable by printer  

### Expected results

File saves with .GCODE extension 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I did slice the same file in windows with Cura 4.8.0 and all worked as expected

GregValiant, 2023-05-17:
Thanks for the report.
Do you have any MarketPlace plugins running?
A Project file would be helpful.

DallasO, 2023-05-20:
Been using Cura for about 5 months and it does not automatically save with .gcode or .3mf extensions.

flatpak
gnome 42.5
popOS! 22.04

closed 2023-06-01 by github-actions[bot]