Ultimaker/Cura #15548: Slice fails and requests to report a bug

Johann1, 2023-05-17:
### Application Version

5.3.1

### Platform

Windows 10 Home

### Printer

Ultimaker S5

### Reproduction steps

1. Open Download Page [https://www.thingiverse.com/thing:4063612/files](https://www.thingiverse.com/thing:4063612/files) and  download Code2_Ulvheim_Tavern_A_Columns_Update.stl or unzip attached project zip file (contains .3mf)
2. Open downloaded STL in fresh installed Ultimaker Cura 5.3.1 or open .3mf project file
3. Start slicing

### Actual results

**Slicing failed with unexpected error and Cura requests to report a bug.** 

[MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [964]: Backend exited abnormally with return code 3221225477!

Screenshot:
[image]

Logs:
[cura.log](https://github.com/Ultimaker/Cura/files/11502142/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/11501775/stderr.log)

Project file in ZIP
[UMS5_Code2_Ulvheim_Tavern_A_Columns_Update.zip](https://github.com/Ultimaker/Cura/files/11501998/UMS5_Code2_Ulvheim_Tavern_A_Columns_Update.zip)

### Expected results

**Slicing works.**

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11502389/cura.log)
[UMS5_Code2_Ulvheim_Tavern_A_Columns_Update.zip](https://github.com/Ultimaker/Cura/files/11502413/UMS5_Code2_Ulvheim_Tavern_A_Columns_Update.zip)
[stderr.log](https://github.com/Ultimaker/Cura/files/11502415/stderr.log)

GregValiant, 2023-05-18:
Thanks for the report and the files.
**Backend exited abnormally with return code 3221225477!** I believe that is a memory error.
I'll mark this as a duplicate and close it as there are a lot of reports on this behavior and that error code.  I hope you understand.
For myself (Windows10 home), I've never been able to duplicate that error.
With your settings this is a 44 hour print.  I didn't find any errors in the model.
[image]

closed 2023-05-18 by GregValiant

Johann1, 2023-05-18:
Thank you for the quick answer. It would make sense to assume a memory issue.
I will try again on a Ti5 Aegis, because i know it has 128GB.

Johann1, 2023-05-19:
@GregValiant - i tried with 128 GB machine but same issue occured after i imported the project and started the slicing:
[image]

@GregValiant - Does it have to do with Ultimaker printer? I have the same error for S3 and S5
> 2023-05-17 22:04:58,846 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [115]: [Backend] [2023-05-17 22:04:58.845] [warning] Failing to discretize parabola! Must add an apex or one of the endpoints.
2023-05-17 22:05:02,994 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [964]: Backend exited abnormally with return code 3221225477!
2023-05-17 22:05:02,997 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [360]: Attempting to kill the engine process
2023-05-17 22:05:02,998 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [366]: Killing engine process
2023-05-17 22:05:02,999 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [369]: Engine process is killed. Received return code 3221225477

I will continue to try other settings next.

Johann1, 2023-05-19:
@GregValiant - It works with 0.15mm  but not with 0.2mm fast print 😒
[image]

GregValiant, 2023-05-19:
An alternative may be to change the Initial Layer Height by just a bit.  For example if it is 0.20 make it 0.21.  Sometimes that can be enough to full the bug and you still get the quicker print time from the higher layer height.  That's a long print and anything to make it quicker will help.
Another thing you could do would be to make the "Infill Layer Thickness" 2X layer height.  It will go down every other layer and (depending on the geometry of the model) make for a faster print.

Johann1, 2023-05-19:
@GregValiant actually it works with 0.201mm (but not 0.21mm). Thank you very much! This solution is fine 🙂