Ultimaker/Cura #15549: Cura AppData>Roaming>cura

infowire, 2023-05-17:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

Ender 3 v2 Neo

### Reproduction steps

When I install Cura I open it up setup (Ender 3 2v Neo) my printer. It seems OK, When I restarted again it freezes right after the loading screen. If I delete profile, AppData>Roaming>cura then I can start the process again... But it's going in the loop...

I keep deleting the profile folder, all starts again. 

### Actual results

I disable my Norton.

I even make a new account in windows.

I have tried uninstalling.

### Expected results

Freeze 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/11502491/cura.log)

No plugins. No sign up.

infowire, 2023-05-17:
Event viewer:
Faulting application name: CuraEngine.exe, version: [1.0.0.0](https://1.0.0.0/), time stamp: 0x643fb550
Faulting module name: ucrtbase.dll, version: 10.0.22621.608, time stamp: 0xf5fc15a3
Exception code: 0xc0000409
Fault offset: 0x000000000007f61e
Faulting process id: 0x0x1840
Faulting application start time: 0x0x1D988D152ED9F1B
Faulting application path: C:\Program Files\UltiMaker Cura 5.3.1\CuraEngine.exe
Faulting module path: C:\WINDOWS\System32\ucrtbase.dll
Report Id: 6312d524-0831-4562-9f6f-504051d87c25
Faulting package full name:
Faulting package-relative application ID:

MariMakes, 2023-05-19:
Hey @infowire,

Welcome to the UltiMaker Cura Github 🚀 
When I read your issues there are two solutions that jump to mind. 

1. Are you trying to do a clean install? Could it be that you are not doing it correctly and that Cura is confused about the missing files? 

> This is how you perform a clean install
> 	1. Quit the application
> 	2. In a file explorer window, navigate to:
> 			My Computer/users/[user_name]/AppData/Local/cura
> 			My computer/Users//AppData/Roaming/cura
> 	1. Delete the entire 'cura' folder of your current version. Note: Cura folders of previous versions will be imported if not deleted!
> 	2. Open Ultimaker Cura. If it has been reset correctly, you will be greeted with the onboarding flow
> 	
> If you do this you will use any custom profiles. If you want to keep these make sure to make a backup.
> You can read more about them here: https://support.makerbot.com/s/article/1667412704080
> 

2. It could also be that you are running into a slowness bug that is reported to be resolved in this special release. You can find the download links and the discussion here: https://github.com/Ultimaker/Cura/discussions/15098

Can you try these solution directions and let us know how they work out for you? 
If it's not the case, just let us know and I'll take another look at what is going wrong. 👍

infowire, 2023-05-20:
After restart the computer, it went back to freezing.

MariMakes, 2023-06-05:
Hey @infowire,

Did you get a chance to test our special release to solve slowness issues? 
You can find the download links and the discussion here: https://github.com/Ultimaker/Cura/discussions/15098

Or did a proper clean install help you?

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker. Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-06-20 by github-actions[bot]