Ultimaker/Cura #15550: Material settings not being applied when a new material selected.

tlhintoq, 2023-05-17:
### Application Version

5.3.1

### Platform

Windows10x64, 21H2

### Printer

All - Not printer specific

### Reproduction steps

Bug discussed with @nallath on Facebook:
https://www.facebook.com/groups/1959877270949115/posts/3486594058277421
And at his request this bug is being created.

Let's say you have a machine like... Ender3pro or Elegoo Neptune3max... Tenlog D3 IDEX... CR10s family... anything on my racks really...
You start making profiles for different use cases:
[image]

- Armor profile for big items with large layer height and thick line width
- FastDraft profile for thin walls, high speed where you don't care about ghosting because its just a draft
- HighQuality where you sacrifice speed for customer-grade finish and smooth surface

But you have different materials.  Esun PETG at 225° with 1.5mm retraction and 0% fan.  Jayo PETG at 230° with 2mm retraction and 40% fan.  Blahblah PLAPLUS at 215° at 2mm retraction and 100% fan.  All values you set in the material definition.
[image]

You place your model, select a profile such as 'Armor', then select the material you're going to use like "Esun PETG". You expect that you'll get what you picked in the order you picked it.

1. First you'd get all the values that go with the profile.  That's the majority of print related items most of which are not material related, like "Outside before inside" on walls... and 5 top layers... Ironing on/off... etc.
2.  When you select a material the user expects the values from the material will come in: Temp, retraction, fan.
Only they don't.  You can change materials 20 times... Materials that are linked to other materials, or materials that have been unlinked makes no difference.  As you can see in the above screen shot, after changing materials the values aren't brought in.  This material is defined at 205/65 yet after selecting this material the job is still showing 235/80

### Actual results

Material values ignored when changing materials.

### Expected results

Material values should come in (over write) when a new material is selected.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://www.facebook.com/groups/1959877270949115/posts/3486594058277421

MariMakes, 2023-05-19:
Hey @tlhintoq,

Thanks for your report 👍
How a setting get's it's definition is actually quite complex. 
It can come from your Printer definition, your Material, custom profiles, or a number of other things. 
[There are some explanations available on our wiki. ](https://github.com/Ultimaker/Cura/wiki/Getting-A-Setting-Value)

To see if cura is showing you the correct value, I'll need your project file.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

tlhintoq, 2023-05-20:
Thanks.  I put a project together for ya that uses some generic files and not customer files that I don't have permission to distribute.

MariMakes, 2023-05-22:
Hey @tlhintoq,

Generic files should be fine, no need to distribute files that you don't have permission to share 👍 

Can you double-check if you shared the project file correctly? 
I can only find the links to the Facebook thread you shared. 🤔 

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.