Ultimaker/Cura #15552: Skirt not taken into account when printing one at a time

konvoj, 2023-05-18:
### Application Version

5.1

### Platform

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Hello my dear friends and ex-colleagues, hi to all of you! We have a small bug here. Project files of course attached.

1. I placed the objects to fit within the volumes of restriction for the printing "one at a time" as usual
2. Then I increased the skirt distance from default to higher values

### Actual results

When the skirt distance is increased the zones of restricted volume needed for "one at a time" didn't reflect any changes in the skirt distance; This will inevitably cause the printhead to collide with the previous print while printing skirt of the next object.
[image]

### Expected results

Skirt distance increasing increases the exclusion volume around object so that there's no possibility of printhead colliding due to this.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Bug_UMS5_TPU Foot Desk.zip](https://github.com/Ultimaker/Cura/files/11505735/Bug_UMS5_TPU.Foot.Desk.zip)

MariMakes, 2023-05-19:
Hey @konvoj,

Thanks for your report 👍

Good find, I recognize your issue!
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14382
We have a ticket on our backlog with the intent to improve this behavior.
We are looking for proof that it fails in practice and not just in theory. 
Would you mind sharing a picture of a printjob that fails because of this behavior?

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/14382

closed 2023-05-19 by MariMakes

konvoj, 2023-05-23:
In practice, nobody uses larger skirt distance than some 5 mm, so it is probably within the margins of printhead exclusion zones as defined now. So this is really just a wrong behavior that probably never shows in real life, since it's a niche on a niche.
I can make you pictures of intentionally failed printjob from the printer, but I understood the issue soon enough not to generate such a printjob in the first place, so I don't have it at hand.