Ultimaker/Cura #15553: Gcode shows: FLOW_RATE_COMPENSATED_OFFSET = -nan(ind) and printer stop after this point

hainguyents13, 2023-05-18:
### Application Version

5.3.1

### Platform

Windows 11 Home

### Printer

Anycubic Vyper

### Reproduction steps

Simply import and slice

### Actual results

Gcode file shows: 
`;TIME_ELAPSED:1057.420577
;LAYER:49
;TYPE:WALL-OUTER
;MESH:[a]_wire_cover.stl
;FLOW_RATE_COMPENSATED_OFFSET = -nan(ind)
G1 F-NAN(IND) X128.369 Y142.758 E-NAN(IND)
;FLOW_RATE_COMPENSATED_OFFSET = -nan(ind)
G1 F-NAN(IND) X128.555 Y142.8 E-NAN(IND)
;FLOW_RATE_COMPENSATED_OFFSET = -nan(ind)
G1 F-NAN(IND) X129.135 Y142.8 E-NAN(IND)`
and printer stop at this point

### Expected results

should generate valid gcode

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11506223/cura.log)
[AV_[a]_wire_cover.zip](https://github.com/Ultimaker/Cura/files/11506232/AV_.a._wire_cover.zip)

GregValiant, 2023-05-18:
Thanks for the report and the files.
This is a duplicate of #15342 .
In the Cooling section change the Minimum Speed to something greater than "0".
The bug is that 0 should be illegal in the box.  Those are very small layers so you are bouncing off the Minimum Layer Time and the Minimum Speed is coming into play.

hainguyents13, 2023-05-18:
> Thanks for the report and the files. This is a duplicate of #15342 . In the Cooling section change the Minimum Speed to something greater than "0". The bug is that 0 should be illegal in the box. [...]

that's it! thank you for the instance reply!

closed 2023-05-18 by hainguyents13