Ultimaker/Cura #15555: 3D Print Not Slicing

JayWhelan1, 2023-05-18:
### Application Version

5.1.1

### Platform

Windows 10

### Printer

Ultimaker s5

### Reproduction steps

Loaded model
orientated on the build plate
Pressed slice

### Actual results

Error code appears
Slicing failed with an unexpected error please consider reporting a bug on our issue tracker

### Expected results

Slicing should have processed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[UMS5_Machine part 3d print.zip](https://github.com/Ultimaker/Cura/files/11507641/UMS5_Machine.part.3d.print.zip)

GregValiant, 2023-05-18:
Thanks for the report and the file.
This sliced fine for me so I'm unable to duplicate the problem.  I'll mark this as a slicing error and the Cura team will take a look.

As a side note I checked the model and it's...odd.  Both MS 3D Builder and formware.co said there were no errors, but here I have the model as "altered" by the formware repair utility on the left (it added triangles and vertices) and the right model is the version included in the project file.
Cura typically will show "flipped normals" in red as an indicator that the surface is facing inside out.  You can see that although the model is watertight there appears to be issues that might be fuzzing the bug.
[image]

Here you can see that they sliced for me and they do look identical.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic