Ultimaker/Cura #15557: (5.3.1) Short Infills between walls

MathiasLHP, 2023-05-18:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Ender 3 S1 Pro

### Reproduction steps

I sliced a simple clip from thingiverse. 
2x wall line count, 50% infill, 0,2mm layer height, PLA

### Actual results

The slicer is creating short infills between wall, when not necessary. Only happening on one side though. 
The workaround is 3x wall line count, but should not be necessary. 
Also with 2x wall line count, it is making some weird transition in the top layers. 

The setting for filter out tiny gaps (or named like) was removed, and so was fill gaps between walls, but it might have worked here?? 

[image]
[image]

### Expected results

I would assume i could disable short infill lines or filter the tiny gaps, or force a thick line instead. I know this feature is available in other slicers. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Long_Slide_Clip_-_175mm.zip](https://github.com/Ultimaker/Cura/files/11510981/Long_Slide_Clip_-_175mm.zip)

GregValiant, 2023-05-19:
Thanks for the report and the file.
I think this is a settiing issue.
If you enable "Connect Infill Support Lines" the problem appears to go away.
Alternatively, if you push your Wall Line Width to 0.5 and make the Minimum Line Width 0.4 this will print without all those extra infill areas.
I'll leave the bug label on this for now, but it does look like there are settings that can eliminate the problem.  Let us know if that worked for you.

In the image below the only change I made to your settings was to enable "Connect Infill Support Lines".  That second Area of Interest also went away.
[image]

MathiasLHP, 2023-05-20:
Hey. Thanks for the quick response. 

I agree that one can manipulate the settings and force the infill away. Although I didn't quick get the results I was looking for, with only Wall Line Width to 0.5 (can't find the minimum line width)

[image]

In regards to Connect Infill Lines, it also works, but ends up with many extra lines. 

I am still counting on a bug, here's why. 

- It works on one side. 
- It works in the old Creality slicer (the setting for Fill Gaps between Walls works flawlessly. I know it was removed with Curas update, since you improved the slicer, hence why I would consider this a bug)
- Also could not reproduce artifacts on top layers in Creality slicer. 
[image]

MathiasLHP, 2023-05-20:
Update: After reading a reddit post about Cura 5 updates, and issues with a model, I tried changing Wall Transitioning Threshold Angle from 10 to 30 deg. 
https://www.reddit.com/r/fosscad/comments/vlfg2y/warning_to_cura_users_cura_5_inexplicably_removed/

I don't know why it worked, maybe mesh related? 

But it didn't work for top layer artifacts. See 2nd picture.  

[image]
[image]

GregValiant, 2023-05-20:
I always check the models first and I didn't find any problems with your model.  When I made my changes that issue went away as well.
I think the radius is coming into play as Cura wants to maintain the constant thickness of the outer part but the slices are planar as they go through the areas of the radii.
I still think there are settings in place to tune the print, but it isn't terribly obvious what needs to be done.