Ultimaker/Cura #15558: Skirt start from inside to outside on Cura 5.3.1 for Mac

StmLord, 2023-05-18:
### Application Version

5.3.1

### Platform

Mac

### Printer

Custom

### Reproduction steps

whenever I try to add skirt to a print it starts from inside to outside and  because I use it as a simpler brim it messes the first layer in the start position

### Actual results

skirt starts from inside to outside

### Expected results

Skirt must start from outside to inside like brim does.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no other info

MariMakes, 2023-05-19:
Hey @StmLord,

Welcome to the UltiMaker Cura Github 🚀 
What an interesting report 🤔 

If I'm reading this correctly you are trying to use a skirt for a different purpose than priming. 
I'm super curious, do you have screenshots of what you are seeing? Because in my case the ordering is a bit random:
[image]

Or even better, do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

StmLord, 2023-05-24:
Exactly, I'm using the skirt as a simpler type of brim, so it will not be as invasive as brim to internal curves, as most of the time build plate adhesion is needed on external corners and not on internal curves. That way is making easier to clean it and keep the ends sharp.
I have included a project file with an example that you can see that skirt is starting from its inside perimeter first, opposite of what brim does or any other older versions pre-5.
Thanks for your time

MariMakes, 2023-06-05:
Hey @StmLord,

Thanks for the project file that made the issue much clearer. 👍 

I'll have to double-check with the team but I think this feature for 5.4 might help your case. https://github.com/Ultimaker/CuraEngine/pull/1862

I expect that if it's not a quick fix that it will not get a high priority to implement a fix, because it's not the intended use for a skirt.  Fingers crossed 🤞