Ultimaker/Cura #15564: M600 error retracting filament

JuanManuelCuello, 2023-05-19:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

Hello, I had Cura 5.2.2 and I have updated to cura 5.3.1. I have noticed that in the new version of cura the change filament function in the g-code writes the m600 code differently which causes it to not retract the filament. As if it did in the previous version.
If I remove the line jump manually in the print, it retracts the filament when activating the m600 command. There is a way that the new version of cura generates the line of m600 equal to the previous version?
Cura 5.3.1
;BEGIN FilamentChange plugin
M600
E30.00 U300.00 X0.00 Y0.00;END FilamentChange plugin
;LAYER:1
--------------------------------------------------------------
Cura 5.2.2
M600 E30.00 U300.00 X0.00 Y0.00 ; Generated by FilamentChange plugin
;LAYER:1
---------------------------------------------

### Actual results

.

### Expected results

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

MariMakes, 2023-05-19:
Hey @JuanManuelCuello,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

Good find! We did introduce some changes in 5.3 like: https://github.com/Ultimaker/Cura/commit/ac16659829eeecccb8ce734bb353dcf4a4f630b6 so it could very well be that this one slipped through.

I'll bring it up to the team to see what they can do to improve it.
Fingers crossed 🤞

GregValiant, 2023-05-19:
I fixed this file.  There was an unwanted newline character after M600.
Unzip it and over-write the file in "C:\Program Files\UltiMaker Cura 5.3.1\share\cura\plugins\PostProcessingPlugin\scripts"
[FilamentChange.zip](https://github.com/Ultimaker/Cura/files/11518570/FilamentChange.zip)

JuanManuelCuello, 2023-05-19:
Thanks, I found the error in line 202, there was a \n after the M600
Excellent service :)

MariMakes, 2023-05-22:
[image]

Awesomesauce 😄 

If https://github.com/Ultimaker/Cura/pull/15581 passes review you can expect a fix in the next stable version of Cura. 
Again, Thanks for the report and excellent detective skills 👍

closed 2023-05-23 by casperlamboo