Ultimaker/Cura #15566: One at a time printing initial layer temperature

AndyShap2020, 2023-05-19:
### Is your feature request related to a problem?

I think I have found a new issue for Cura. I'm using 5.3.1.
I use one at a time printing.  On the particular parts I print, they print better one at a time and if the power goes down (it does all the time!) I only lose the part I was printing, not the whole batch.
My problem is that I have noticed that some of my prints are warping on the bottom.  I have realized that although I am set to a 65C initial layer and 60C all other layers, this is only for the first print! After that all the other prints stay at 60C for their first layer.  I'd think if Cura was going to offer OAAT printing, they would adjust the G Code to set the temp higher for each first layer.

### Describe the solution you'd like

I'd like it to go to 65C for the intial layer, then to 60C for remaining layers. Then, when the second print starts, got up to 65C then down to 60C, and repeat for all other prints in this OAAT print job.

### Describe alternatives you've considered

`

### Affected users and/or printers

I would think that any initial layer settings in Cura should apply to the initial layer of each OAAT print.

### Additional information & file uploads

`

GregValiant, 2023-05-19:
"I would think that any initial layer settings in Cura should apply to the initial layer of each OAAT print."
I agree.  It will probably be a bit before they get to that.

In the meantime you can try this.
In Cura - Set both your build plate temperature and initial build plate temperature the same. (60 or 65...it doesn't matter).
Add 2 instances of the "Search and Replace" post-processor ("Extensions/post-processing/Modify Gcode/Add a Script"

The first Search and Replace will be:
Search:  ;LAYER:0\n
Replace: ;LAYER:0\nM190 S65\n
Use Regular Expressions:  Checked

The second Search and Replace will be:
Search:  ;LAYER:1\n
Replace: ;LAYER:1\nM140 S60\n
Use Regular Expressions:  Checked

The first search will replace all the LAYER:0 lines with two lines
;LAYER:0
M190 S65
That will stop printing until the bed hits 65.  You could also use an M140 and there would be no waiting.

The second search and replace will replace all the LAYER:1 lines with
;LAYER:1
M140 S60

"\n" is the Python newline character.  They are required as shown.  Make sure you get your spelling correct as Regex is fussy and you can get unexpected results.

AndyShap2020, 2023-05-19:
That's great! I'll give it a try. Thank you.

GregValiant, 2023-11-02:
This appears to be fixed in 5.5.0.  Can we close this?

AndyShap2020, 2023-11-02:
Yes. Can you give me some guidance on how to use it in 5.5.0?

GregValiant, 2023-11-03:
If the "Build Plate Temperature Initial Layer" is different than the regular "Build Plate Temperature" setting then Cura will make the adjustment at model change.
I set up a slice using OAAT with the Initial Layer Print Temperature at 210 (regular at 205) and the Initial Layer Bed Temperature at 60 (regular at 50).  I get this at model change:
M107 ;fan off
G0 F9000 X11.307 Y12.451 Z105
G0 X193.788 Y12.911
M140 S60 ;heat the bed to Initial Layer Temperature
M104 S210 ;heat the hot end
M105 ;temperature report
M109 S210 ;wait for the hot end
;LAYER_COUNT:500
;LAYER:0
Then at the second layer:
;LAYER:1
M140 S50
The hot end temperature is adjusted a few lines ahead because _when_ it is inserted into the gcode is controlled by the setting "Cool Down Speed.

I'll go ahead and close this as it appears to be fixed.  If you are still having trouble you can still respond here and I'll get notified.

closed 2023-11-03 by GregValiant

AndyShap2020, 2023-11-04:
Thanks Greg, I’ll try it out. I appreciate the work you’ve done!

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>

AndyShap2020, 2023-11-06:
It works great! Thank you!

_______________________________________________________________

Andy Shapiro, Owner
AS Consulting  (Networking, Hardware, Software & Training)
P.O. Box 1728, Goleta, CA. 93116-1728
Voice: (805) 968-6366
email: ***@***.***
Web: www.asconsulting.com<http://www.asconsulting.com/>