Ultimaker/Cura #15567: app crashing

CarrieSimba72, 2023-05-19:
### Application Version

5.3.0

### Platform

Windows 11 I think?

### Printer

Anycubic Kobra Neo

### Reproduction steps

resizing, moving, etc

### Actual results

[image]

screen goes white & freezes. just trying to move or rescale something. Sometimes it works sometimes it doesnt. I try to resize & I have to guess if whatever move I make makes it large or small. and sometimes it makes them too big. sometimes it works fine. I often have to press ctrl + alt+ delete & shut the program down or restart the computer. Im on a gaming laptop.

### Expected results

when it does work i can actually get files up & going. idk why it keeps crashing? It does that with any file I am trying to do this with. It's not just 1.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

are there any current updates?

MariMakes, 2023-05-22:
Hey @CarrieSimba72,

Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13820
More people reported slowness with Cura and Windows 11. I personally upgraded my Windows to 11 to reproduce, but I haven't experienced it. 

We might have a fix for this issue. It's a bit hard for us to test, but we have a nightly build available.
Can you please try it out and let us know if it resolves your issue?
You can find the download links and the discussion here: https://github.com/Ultimaker/Cura/discussions/15098

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13820

closed 2023-05-22 by MariMakes

CarrieSimba72, 2023-05-23:
worked better. thank you