Ultimaker/Cura #15570: Gradual infill creates very short line segments (wastes time & impact quality)

carribeiro, 2023-05-20:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

I'm printing a large vacuum-forming mold that will take nearly 2 days to print. As I intend to fill it later with resin (to make it solid) I usually use no bottom layer and a pretty sparse infill. I'm using progressive infill so I added gradual infill steps.

When NOT using the "Connect Infill Lines" option, I noticed that the slicer creates tiny segments as if the infill was already at the highest density. At first, I thought it happened on inclined walls but then I noticed that it also happened when the infill touches straight walls.

[image]

As you can see there's ton of very small line segments which cost HOURS of extra printing time (4 hours more in this case). It also creates more artifacts and impacts quality.

When using the "Connect Infill Lines" setting, it eliminates a lot of these issues, and prints noticeably faster, but I found other possible problems.

[image]

- On the external wall the artifacts disappeared, but around the internal pocket of the mold the behavior is a bit erratic. 
- On the first layer of each density doubling, the algorithm is "connecting" the infill lines over nothing. Of course this is unavoidable (unless an improvement is made which I will suggest later), but I assume that if there's no wall, it shouldn't be connecting the line either.

[image]

Not only this but consecutive layers don't even have the same "connections".

[image]
[image]

A POSSIBLE IMPROVEMENT

I know you probably have a very large list of things to do but I think that the gradual infill steps would work better if the transition to a higher density extended the lines over the entire span of the cells of the previous layer. This doesn't seem difficult to do with grid or lines infill, which I assume that are the easier ones to use with gradual infill anyway. is it too much to ask for such a feature?

### Actual results

If I don't use the "Connect Infill Lines" setting, the slicer generates lots of very small segments which cost time and impact quality.

If I use the "Connect Infill Lines" setting, the quality is improved but the end result is a bit more erratic over several consecutive layers. I'm testing the result now.

### Expected results

In the ideal case, the infill shouldn't require the "connect infill lines" to generate a clean print. Also, the transition to a new density should make sure that the first layer would be better supported by briding the "cells" of the infill of the previous density.bridging

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I've not added the file because it's a customer project. I may be able to create a sample model if necessary but the screenshots are very clear, and I'd like to know if this is a report worth analyzing before wasting more of everyone's time :-)

MariMakes, 2023-05-22:
Hey @carribeiro,

Thanks for your report 👍
Before we dig into the bug that you are reporting, is your goal to have a speedy print with minimal infill? 

Because I believe that can also be solved with Lightning Infill and Skin Edge Support Layer settings. 😉 
[image]

I totally understand if you can't share the model, so I made a similar one we can troubleshoot with. 
[Mari_CE3E3V2_Bunny_Side_1.zip](https://github.com/Ultimaker/Cura/files/11531585/Mari_CE3E3V2_Bunny_Side_1.zip)
Can you take a look if this solves your problem? 

If you would still like us to resolve your original problem, would you mind sharing and updating the shared bunny project with the settings that you are using? Then we can use that project for troubleshooting purposes. 

Thank you 😊

carribeiro, 2023-05-23:
I like the lightning infill, but in my experience, it not always produces parts that are as strong as a line or grid infill with the progressive infill.

To understand why, you must understand our requirements. It may look a bit too "particular" to my case, but I believe this is useful in an increasingly important scenario, that is the production of molds for other industrial processes.

I'm producing molds for vacuum forming (and will be producing mold for other processes too). PLA HT is strong enough to withstand the pressure and the temperature involved for small-scale production (< 100 parts) but requires a SOLID model to survive several cycles of vacuum. Printing 100% infill is clearly impossible, so I need an infill that is both strong, but also "fillable" with some other material after being printed.

The lighting infill tends to produce a very complex geometry that traps air and makes it harder to fill with resin. It's also prone to problems with some parts of the infill disconnecting from the mesh and (possibly) getting loose within the part.

By using the progressive infill, I can minimize the infill and the printing time, but more importantly, the resulting structure has large enough holes (and always open at the top) to be filled with conventional resin (polyester or similar). 

Also, the progressive infill produces a grid that helps to keep the overall shape of the model  by "locking" the side walls in place.

carribeiro, 2023-05-31:
Hi, Just to let you know I'm still working on it.

MariMakes, 2023-06-05:
[image]

Have you considered using Infill Support? 
It only has very incidental applications, but it might fit yours.

carribeiro, 2023-06-12:
I apologize for the delay to supply more information. But I think I've build a very simple case to demonstrate both the problem (which I think is really a bug, something to be corrected) and also something that may be a feature request in the future.

The goal is to print a part for vacuum forming. Vacuum-forming molds need to be very strong and making them 100% solid is a frequent recommendation. However, to print 100% infill is both too time consuming and expensive. So our alternative is to print the part OPEN AT THE BOTTOM and with very little infill, which makes it to print MUCH faster, and use MUCH LESS material. The model can later be filled using some less expensive material; for example, a mix of quartz (or sand) and resin; or alternatively plaster. This filling is strong and saves material and printing time (in the case of plaster, it takes too long to cure in our experience, but that's not the point of this issue).

The test part is a very simple cone.

[image]

We slice the part using GRADUAL INFILL and THREE steps. This way the bottom part is very sparse and allows the filler to flow to the bottom of the mold (which is the top of the molded part).

After doing it, notice how there are several "micro lines" at the periphery. Such lines cause several problems and impact both the printing time and the quality of the part.

[image]

[image]

A perspective view shows how much time and material is spent over several layers and the artifacts it generates;

[image]

The situation doesn't improve after the infill doubles in density.

[image]

If we select CONNECT INFILL LINES, the situation improves a bit. The connection lines turn to be an extra inside line; the files takes a little longer to print but with less artifacts and better quality.

[image]

[image]

This bug request focus only on the strange lines that are created inside. I guess that if I could remove any lines from the infill which were shorter than some value (like, smaller than 5mm, and connected to the inside perimeter only from one side, maybe I would get what I want. But it's clearly an artifact that I believe should be looked at.

carribeiro, 2023-06-12:
Here follows the 3MF (zipped) file so you can reproduce the settings.

[CE3E3V2_Test Cone.zip](https://github.com/Ultimaker/Cura/files/11728741/CE3E3V2_Test.Cone.zip)

carribeiro, 2023-06-13:
I think I understood why these lines show up, and I believe that I realize why it's probably hard to "fix" in a definitive way.

The problem is: slightly inclined faces have regions which are under the "Gradual Infill Step Height" setting, but where the lines are clearly not meaningful. 

My best guess is that the algorithm should make sure that the gradual infill would only kick in when a large enough AREA is under the limit. It could a percentage of the total area, for example. This would probably have other side effects, but it's a start.

carribeiro, 2023-06-24:
One more example where Gradual infill fails badly. In this case I can use the lightning infill, but it's clear that Gradual Infill doesn't produce functional results for many parts. I guess the algorithm needs a lot of adjustments and that it may not be easy, but I still like the idea, it seems more regular and predictable than lightning for some use cases.

(in this case the gradual infill, even at lower percentages, still adds some stiffness to the bottom of the box, and allows us to use fewer bottom layers)

**THE MODEL**

[image]

**THE INFILL RIGHT AFTER THE TRANSITION UNDER THE LETTERS**

[image]

**DETAIL**

[image]

[image]

Note how the small infill segments still appear at the walls, even if the will is a straight one. That's clearly not the way it should be.

[image]