Ultimaker/Cura #15571: (5.3.1) Multiple model project draws skirt around a subset of models and only starts printing the other models at layer 8

GusMcN, 2023-05-20:
### Application Version

5.3.1

### Platform

Window 10

### Printer

Ender 3v2

### Reproduction steps

1. Created a project containing 4 copies of a model of an M6 nut and 6 copies of an M6 bolt and sliced it.
2. In the output G-code the skirt is drawn around the nuts only and the bolts are ignored until layer 8 which then attempts to print them in mid-air. If I'd looked at the preview I would have realised there was a problem.
3. Project file available at https://www.dropbox.com/s/wazzj1b12ivqmrf/strut1.3mf?dl=0 will demonstrate the problem.
4. Attached screengrabs show Layer7 and Layer8 previews
[image]
[image]

### Actual results

Abtsract 3D sculpture

### Expected results

Some nuts and bolts

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

. Project file available at https://www.dropbox.com/s/wazzj1b12ivqmrf/strut1.3mf?dl=0 will demonstrate the problem.

[image]
[image]

GusMcN, 2023-05-20:
Pretty certain that this was caused by a tangency constraint in the model causing the threads to create a series of points at z=0 on the bolts and then nothing until the flat face higher up. The flat face should have been at z=0.
Sorry for the trouble.

Asterchades, 2023-05-21:
Yep. If you zoom right in on the Prepare screen, and angle up from underneath to make it easier to see, there's a series of small, perfectly flat squares touching the build plate.

Going to close this one.

closed 2023-05-21 by Asterchades