Ultimaker/Cura #15572: Need option to NOT update Printer Settings from settings saved in .3mf project file

MadFDMer, 2023-05-21:
### Is your feature request related to a problem?

I have lot's of project files, which save the printer settings. But I update my printer settings every so often, for good reason. So when I open an old project file I want to use my updated printer settings, and I DON'T want the project file to automatically overwrite my printer settings with the old printer settings saved in the project file...and I DON'T want to create a new printer either. Also, I may have multiple printers, or I may have 1 printer setup with 2 or more printer profiles...and so same issue, when I open a project file, I just want to use one of the current printers I have setup, and not overwrite it with whatever printer settings are saved in project file. It is SUPER annoying that it updates the printer settings when opening a project file, and it doesn't even tell you that it is doing it.... except for the little i-nfo button does mention it.

### Describe the solution you'd like

When opening a project file, should have a prompt/option in the Open Project summary page to have a check box next to the  Printer settings Open with dropdown, that if checked, it will NOT update the the Printer Settings from the project file.

Alternatively, Printer Settings could be made just like Print Settings profiles, as in, you can modify settings, but they are not permanent until you save the profile. So Cura could open a project file and update the printer settings from the project file (like it does now), but if I don't like that, there should be a restore to Defaults button like Print Setting profiles have, so I can restore my SAVED printer settings from a profile. And, also, obviously, there would be a save button so you could save any changes to the Printer Profile IF YOU LIKE, not if CURA feels like it.

### Describe alternatives you've considered

Have to create a new printer every time, and then switch to the printer profile I want after opening the project file...but that doesn't work either because Cura has a bug and will change the print settings when you switch printers, see bug #15436 "Print Settings change when switching printers". So basically there is no work around right now other than to just rip your hair out and go insane! 

### Affected users and/or printers

All

Asterchades, 2023-05-22:
By default when you open a 3MF project file it will ask you if you want to import the entire project or just the models contained there-in. Importing the models alone will retain all of your existing settings.

Should it not be presenting this dialogue it would be because you, at some point in the past, have told it not to ask you again when selecting one of the two options (it sounds like you chose "Open as Project"). You can change the default option or restore the prompt in the Preferences - General tab, down near the bottom with a setting called "Default behaviour when opening a project file".

MadFDMer, 2023-05-22:
Hi Asterchades,
Yes, I understand that. But that was not not point. Sometimes I have a 3MF file that is older, and when I open it I want to update it with my current printer settings BUT I want to keep the slicer/print settings in the 3MF because they are specific to that project. And sometimes I may want to use the printer settings saved in the 3MF, but typically I don't. I keep my printer settings updated on Cura...and I have multiple instances of the same printer with different start G-codes for  different nozzles, so I really don't want to mess with my printer settings messed with when opening a 3MF file, unless I want to. There should be an option to do both.

mike-brady, 2024-11-24:
I agree with @MadFDMer. My machine settings are specific to how I have my machine set up right now. If I open up an old project I want it to load with the profile it was originally saved with. But it shouldn't be changing my machine settings to however my machine was setup in the past.