Ultimaker/Cura #15574: No ability to paste code, or create second line, in Post Processing Plugin->"Insert at layer change" (recurrence of bug 12381?)

TsetTsyung, 2023-05-21:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Copied layer change code from Heat Tower model (any G-Code should do)
2. Opened Cura.
3. Clicked on "Extensions->Post Processing->Modify G-Code.
4. Selected "Insert at layer change", and selected "Before".
5. Pressed CTRL+V to paste code.

### Actual results

Nothing was pasted into the textfield. No right-click context menu either.

### Expected results

Copied G-Code should have been entered into textfield upon CTRL+V

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The above issue was previously seen as bug 12381 (https://github.com/Ultimaker/Cura/issues/12381}, for version 5.0.0 and was closed as 'fixed'.

Further to the above, I tried to enter the code manually, however I could not enter a new line without the dialog box closing.  Perhaps the read of a newline character is stopping the code from being pasted?

Below is the actual code I was trying to copy/paste was from a heat tower project(https://www.thingiverse.com/thing:2729076).

TsetTsyung, 2023-05-21:
UPDATE:
I deleted all of the newline characters from the code's text file, copied the code and then **WAS** able to paste it into the textfield.  Is this actually a case of "Not a bug, it's a design feature"?

Will attempt print and post results.

GregValiant, 2023-05-21:
Thanks for the report.
There are "newline characters and then there are "newline" characters.  In addition there are textboxes configured for single line display and configured for multi-line display.  Newline characters are often language specific.  Multi-line/Single-line is a property of the box.  Another thing appears to be that those a Prusa scripts and you are attempting to use them in Cura which does not allow math or logic statements in the boxes.
I believe that particular textbox is configured for single-line display.  You cannot copy and paste these two lines together:
HELLO
WORLD
But you could paste them separately as "HELLO" (add \n and paste) "WORLD".
The "copy and paste" attempts to put multiple lines into a box that is not configured for multi-line display OR you are attempting to insert the wrong newline character.
The required newline character is the "\n" Python newline character.

This may be more of a misunderstanding than a bug.  Looking at the Thingiverse page it looks like the statements you are trying to past in are illegal in Cura, but I may be looking at the wrong thing.
Right now I don't see a bug here.  If you could post the code you were trying to paste into the box that would help.

TsetTsyung, 2023-05-21:
Hi GreValiant,

I have a print (4th attempt) going now, but all you have said applies.

I came to the same realisations after I post my previous comment and took a deeper dive into gcode, plugin code and prusa vs cura plugins.

A) I was, indeed, trying to enter prusa code.
B) It does appear that the textfield is single line only.
C) The print running now is using custom editing of GCode to enter temp changes at the right layer changes. I did, sctually, use something like HELLOWORLD to mark where this would go in code so I knew where to include the temp changes.
D) Good to know about the interpretation of newline character strings! I may very well use that in the future.

I would agree that this is definitely a case of my misunderstanding curas post processing functionality. PEBCAK, PICNIC & ID-10-T all apply [facepalm]

I will close this issue now, and apologise for wasting anyone/everyones time.

Thank you so much for clarifying this point for me, GregValiant! You star.

GregValiant, 2023-05-21:
Another thing to consider is that I don't believe the regular "Insert at Layer Change" allows multi-line insertions as it does not have Regular Expressions capability.
I play around a lot with the post processors.  Attached is my interpretation of "Insert at Layer Change" and does allow multi-line insertions when a comma is used as the delimiter.  Still no math though.

Unzip and put the.py file into "C:\Program Files\UltiMaker Cura 5.3.1\share\cura\plugins\PostProcessingPlugin\scripts"
Since the old file and the new file both have the same name you might want to rename the stock "SearchAndReplace.py" file before pasting this one in so the stock file doesn't get over-written.
[InsertAtLayerChange.zip](https://github.com/Ultimaker/Cura/files/11524594/InsertAtLayerChange.zip)

MariMakes, 2023-05-22:
Awesome troubleshooting @GregValiant 💖
Very glad to read your issue has been resolved. 

I'll close this ticket. 👍

closed 2023-05-22 by MariMakes