Ultimaker/Cura #15575: M141 command appearing after start gcode.

rlhahn, 2023-05-21:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Sidewinder X1

### Reproduction steps

added "SET_GCODE_VARIABLE MACRO=START_PRINT VARIABLE=chamber_temp VALUE={build_volume_termperature}" to Start G-code

### Actual results

After Start gcode, cura still adds the M141 S command
[image]

### Expected results

The M141 S command should not be added when {build_volume_termperature} is specified in the Start G-code, like how when {material_print_temperature_layer_0} is specified there is no M109 S command added.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

GregValiant, 2023-05-21:
Thanks for the report.
I don't see a bug here because Cura only speaks Cura.  Whatever language that "macro" is for (Octoprint, Klipper, Bambu, etc.) isn't understood by Cura so you still get the M141 command.
You would need to put the M141 command in your Startup Gcode.
M141 S{build_volume_temperature}
It's the missing M141 that is the problem.  It can go where it is, or a few lines later in the StartUp.  It's going to be there someplace as your macro lines are just gibberish to Cura.

I'll leave the bug label on this for a bit so you can respond but I'm inclined to remove it.

rlhahn, 2023-05-21:
I understand that Cura doesn't understand the line but I would like this to function like M109 does. For ex. if I put noting in my start Gcode, Cura will add a M109 S command but if I specify somewhere {material_print_temperature_layer_0}  in my start Gcode (doesn't have to be specifically a M109 command as you can see from my example) Cura will not put the M109 S command (even tho it is missing from the start Gcode).

GregValiant, 2023-05-21:
I played around some more and whether or not there is an M141 line in the StartUp - Cura will still add the M141 line after the StartUp.
M141 S40
;
;End of StartUp
;
M141 S40
The only way to get rid of it at this point is with Search and Replace
Search: M141
Replace: ;M141
Adding the semi-colon in front will make it a comment and the printer will ignore it.
Here I have removed the line from the StartUp and enabled Search and Replace as described.
G0 F5000 X3 Y50 Z0.35 ;wipe
;
;End of StartUp
;
;M141 S40

rlhahn, 2023-05-21:
Yes this is what I ended up doing as well. Thanks