Ultimaker/Cura #15577: Cura cannot produce circles nor smooth curves rather it produces segmented straight lines poorly mimicking a curve.

DDG127, 2023-05-21:
### Application Version

5.3.1

### Platform

windows 10

### Printer

Tronxy x5sa/x5st/Pro400

### Reproduction steps

Saved a 3d file with the extension .3mf [obj files are even worse], inspected the file and noted RIBS in the curvature areas.  I have attached a png snapshot pointing out the flaws.

### Actual results

When printed these ribs are very pronounced.  How can I get a smooth curve?

### Expected results

Smooth curves, no ribs.  I know that the printer is capable of printing smooth curves.  I cannot get this to work.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Vent cover.zip](https://github.com/Ultimaker/Cura/files/11525589/Vent.cover.zip)
[image]

GregValiant, 2023-05-21:
Thanks for the report.
There are no circles if a computer is involved.  PI has been calculated to 30 trillion decimal places so there are always chords.  
The facets you see on a model are from the translation into STL or 3mf, OBJ, etc. from the software the model was developed in.  You can increase the resolution within the CAD-to-STL translation software but only up to a point.  Cura just gets the model as it exists.  If the model is low resolution there are large facets.  If it is high resolution there are smaller facets.  The G2 and G3 commands turn circles over to the firmware.  Marlin has a setting for "Minimum Arc Segment Size" that sets the limit for each line segment that makes up (what looks like) a circle or radius.

So there is no bug here and I'll remove the label and close this.

closed 2023-05-21 by GregValiant

DDG127, 2023-05-22:
Minimum Arc Segment Size - Is there a setting to adjust in Fusion 360?  I looked again in Fusion 360, I don't see it. 
I have seen 3D printers create perfect circles.

GregValiant, 2023-05-22:
They only look like perfect circles, they are still a bunch of line segments.  The shorter the line segments the more precise the circle.
When you export the model from Fusion there should be options.  In AutoCad it's "FacetRes" but I don't know Fusion.
The screenshot below shows the same model but the left was exported at hi-res and the right was exported at low res.  You can see that the facets are much larger on the low res model.  The Mesh that each model consists of is made up of triangles.  Triangles don't have curved sides.
[image]

DDG127, 2023-05-22:
Thank you for your help.  I will look into this. and hopefully get better at printing.

DDG127, 2023-06-03:
I'm sorry to bother you again, but I have used my 3D CAD files with another slicer and the curves are smooth and seamless.  I use the same file with CURA and there are obvious segments.  This appears to be a CURA problem and nothing to do with the CAD software as described above.  Please look at this again.

GregValiant, 2023-06-03:
No slicers have the capability to change a models resolution.  It is what it is and is built into the model file.
Below are several pieces of software showing  your Vent model.  The facets are the same in all of them.  No slicer can change that as it can only slice the geometry that it is given.
There is no "AI" built into any slicer that can say "Well this looks like it should be an arc so I'll make it an arc".  All curves are a collection of straight lines.  Circles only exist as polygons with a lot of sides.  How would a slicer be able to decide that your 12 sided dodecahedron should really have 200 sides so it looks more like a circle?  What if you actually wanted the 12 sided print?

This is your vent in Prusaslicer.  
[image]

This is a screenshot from Cura
[image]

This shot is from IdeaMaker
[image]

This one is from MS 3D Builder.  I was able to paint the facets so they are more obvious.
[image]

And this one is from Bambu Studio
[image]

Here is a comparison of two STL's in Cura.  They are of the same model.  The model on the left was exported from AutoCad at the highest resolution I can get and the right model is at low resolution.  The right model is highly faceted because that's the way the model is represented within the STL file.
[image]

Here are the same two models in PrusaSlicer.  The problem is identical.
[image]

I can re-open this if you want, but I really don't see anything that can be done from a slicer point of view.  You can call it a bug if you like, but if fixing the problem is the crux of the matter then I think you need to look at how your models are exporting and what the settings are.

I've been working with CAD software since the late 1970's.  There are tricks you can do in CAD to change the resolution of an export.  There is nothing that can be done to increase a models resolution once the export file is created because the actual model is required to do so and that actual model only resides in the CAD software.