Ultimaker/Cura #15578: Bad slicing procces

FoxMulderX77, 2023-05-21:
### Application Version

5.3.1

### Platform

Windows 10 PRO

### Printer

Ender 3 MAX

### Reproduction steps

I modeled a piece in Autodesk Maya and some other programs and saved it as an STL file

[image]
[image]

### Actual results

Sometimes the model looks fine in the _PREPARE_ tab, but once the program ends its slicing process the _PREVIEW_ tab shows a bad result in some areas and the model is too different and blocky in some areas making the piece unuseful
PREPARE TAB
[image]
[image]
PREVIEW tab
[image]
[image]
[image]

### Expected results

I was waiting for no errors at all

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

https://mega.nz/file/zMRgEDjD#Pu244FJ7ViON2M-ehJdoF2Rtp7bBN_AQ-O2JrxpPrBQ

FoxMulderX77, 2023-05-21:
Sorry if this issue is already in the forums but I didn't find anything

MariMakes, 2023-05-22:
Hey @FoxMulderX77,

Thanks for your report 👍
I believe I've spotted what is going wrong. 
You have the setting _Make Overhang Printable_ enabled.
This setting _changeS the geometry of the printed model such that minimal support is required._
[image]
I believe your model should look as expected if you disable the setting. 
[image]

To me, it doesn't seem like a bug in Ultimaker Cura anymore, but an issue with setting choices. 
Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-05-22 by FoxMulderX77