Ultimaker/Cura #15579: Sign In Failing - Cura 5.3.1 - Win 10

JustinKaseToo, 2023-05-22:
### Application Version

5.3.1

### Platform

Win 10.0.19045

### Printer

Ender 3 Pro

### Reproduction steps

I am unable to Sign In to Cura 5.3.1 - screen locks and blue wheel spins.  It doesn't seem to time out (or maybe I am too impatient and need to wait more than 5 minutes).  However, I can close out the Application.

In the log I see:

> UM.TaskManagement.HttpRequestManager._onRequestError [369]: request[aa5fa77e][get][PyQt6.QtCore.QUrl('https://software.ultimaker.com/latest.json')][timeout=None][no-data] got an QNetworkReplyError Host software.ultimaker.com not found. The server returned: b''

So, it seems like it isn't reaching the site.  I am able to log into the site and Marketplace directly.  I don't have a VPN setup.

I attempted an uninstall/reinstall with no luck.

Thanks, in advance, for any assistance.

[cura.log](https://github.com/Ultimaker/Cura/files/11526802/cura.log)

### Actual results

Screen locks while trying to Sign In.

### Expected results

Sign In would complete.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11526806/cura.log)

GregValiant, 2023-05-22:
Thanks for the report.
I have the same printer.  Did you install yours as a "network" printer?  My E3Pro does not have network connectivity and I have never received a "sign-on" request for either a network or the Marketplace.

JustinKaseToo, 2023-05-22:
I'm having the Cura Sign In issue just on my desktop.

I recently added a Creality Sonic Pad to my E3Pro, so it is now Network Connected, and I was updating Cura from my older 4.3.1 (IIRC) so everything would be current.

I can still export the files manually (on a USB) and the Klipper pad works without issue.  Once I can sign into Cura, I should be able to use the Moonraker plugin to then connect to the pad through the network.