Ultimaker/Cura #15582: (5.3.1) Error 3221225477 Unable to slice anything

dmckerson, 2023-05-22:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

Creality Ender-3 S1

### Reproduction steps

1. Upload gcode file (any file)
2. Attempt to slice
3. Receive an error
4. Rotate the file
5. Error while slicing
6. Keep turning the file and try to slice at every step
7. Some files never slice
8. Try again with a different profile (same result)
[example.zip](https://github.com/Ultimaker/Cura/files/11530794/example.zip)

### Actual results

Receive an unknown 
[error-log.txt](https://github.com/Ultimaker/Cura/files/11530781/error-log.txt)
error message

### Expected results

The file should slice (and some files do)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[example.zip](https://github.com/Ultimaker/Cura/files/11530809/example.zip)
[error-log.txt](https://github.com/Ultimaker/Cura/files/11530810/error-log.txt)

GregValiant, 2023-05-22:
Thanks for the report and the files.
I can duplicate the "3221225477!" memory error on my system.  It did slice once but when I tried to duplicate the steps, I couldn't repeat it.  I didn't find any errors in the model.
This appears to be a slicing error and the Cura team will take a look.
[image]

dmckerson, 2023-05-23:
@GregValiant Thank you for picking this up! Just to set my expectations - how quickly does this kind of issue get resolved and released?

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic