Ultimaker/Cura #15583: Add Spiral Z-Hop and Diagonal Z-Hop like BambuStudio and Orca Slicer

Guaffo, 2023-05-22:
### Is your feature request related to a problem?

With normal Z-Hop the printer produces stringing.

### Describe the solution you'd like

With this modification you should have less stringing without giving up Z-Hop. The nozzle should clean itself by wiping with the object. In the Spiral Z-Hop function, the nozzle goes up as it makes circles creating a spiral.

### Describe alternatives you've considered

/

### Affected users and/or printers

/

### Additional information & file uploads

By spiral Z-Hop I mean this:
[image]

By diagonal Z-Hop I mean this:
[image]

gaby64, 2023-05-30:
I would like an option for this aswell.

TeachingTech made a post processor to achieve this:
https://teachingtechyt.github.io/diagonalZhop.html

A built in option would be much more convinient.

SrGatoo, 2023-06-12:
Spiral-hop (Like does the bambu) would be amazing for managing stringing.

gaby64, 2023-10-02:
I am once again looking for this

gaby64, 2023-10-03:
For anyone interested, I found a post processing script for cura:

https://github.com/5axes/Cura-Postprocessing-Scripts/blob/master/DiagonalZHop.py

NadineMik, 2023-12-08:
Hi guys,
I started working on this, and submitted a pull request for a spiral z-hop which is slightly different from the bamboo version, because it creates an arc which is always tangential to the previous and following movement. However, this merge request only contains the logic for a z-hop at the beginning of a travel and still does a normal unretract and un-hop.
Please let me know if you find any issues with this.