Ultimaker/Cura #15584: Error when slicing at Layer height 0 .12 (works on 0.16)

MikeClayton999, 2023-05-22:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

Ender 3 pro

### Reproduction steps

1. Dowload the following STL https://www.thingiverse.com/thing:6003742
2. Slice Earth.Part.2.stl (0.12mm, 10% infil, Bed 60 initial, Nozzle 200 initial, Bed 50 after fist layer, Nozzle 205 after first layer)
3. Extension, post processing, Modify G-Code add pause at height 1.64.
When slicing the program errors.

If I change the layer height to 1.6 (from 1.2) the STL slices without error.

"Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker"

### Actual results

"Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker"

### Expected results

STL should have sliced.  WIll silce on 0.16 layer height but not on 0.12.  First map Earth.Part.1.stl sliced and printed correctly

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11534748/cura.log)
[CE3PRO_Earth.Part.2.zip](https://github.com/Ultimaker/Cura/files/11534756/CE3PRO_Earth.Part.2.zip)

MikeClayton999, 2023-05-24:
> Thanks for the report and the files.
> Changing the Initial layer Height might fool the bug as well.  That would allow you to keep the higher resolution of the .12 layer height.  (0.14 Initial Layer Height worked for me.)
> This sounds like a duplicate of several other bug reports.  I'll mark it as such and the Cura team will take a look.

Thank you for the tip.  It is printing at the moment (20 hours in, 1.5 to go).  I got it working by selecting adaptive layers, on 0.12 layer height, which by the description of your work around probably did a similar thing.Many thanksMike

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic