Ultimaker/Cura #15589: slicing failed

goofyseeker311, 2023-05-22:
### Application Version

5.3.1

### Platform

win11

### Printer

elegoo neptune 3 pro

### Reproduction steps

trying to slice with default, no supports, adhesion custom raft

### Actual results

slicing failed with an unexpected error

### Expected results

that slicing would work without errors

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

wind turbine project file that slicing fails completely.

[EN3PRO_conewindturbine8-v49.zip](https://github.com/Ultimaker/Cura/files/11537659/EN3PRO_conewindturbine8-v49.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11537665/cura.log)

GregValiant, 2023-05-24:
Thanks for the report and the file.
I always check the models first since it's easy and can save time later.
This is the report from formware.co/OnLineSTLRepair
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 13915 Non-manifold edges (?)
--> 14117 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 12772 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 122574ms ------
-> Vertex count changed from 50818 to 72532 (+21714)
-> Triangle count changed from 116878 to 148098 (+31220)

Looking at the model in Mesh Mixer it appears that ALL of the surfaces are inverted.  When that's the case the model usually can slice and in this case (for me) it did.  This is the un-repaired model.
[image]

I'll leave the bug label on this because I think it should have sliced and I'll mark it as a duplicate slicing error.  The Cura Team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic