Ultimaker/Cura #15591: Cura won't generate support under

dymbox, 2023-05-23:
### Application Version

5.3.1

### Platform

macOS 11

### Printer

FlyingBear Reborn2 (unrelated to printer)

### Reproduction steps

1. Download model from https://www.thingiverse.com/thing:2551559 and open in Cura the "Bottom_3.stl" part
2. Set the infill to like 10 to 20% (whatever pattern), layer height 0.25 mm, enable support everywhere.
3. Slice the model and set the preview Z scroll bar to layer 21

### Actual results

You'll see that Cura will try to build some of the globe's bottoms in the air with no material under it. Actually, the fact that some others are be based on the infill walls is purely random thing, just play with infill settings to check it out.

### Expected results

Support structure under the globes' bottom, or infill pattern optimization... Whatever to avoid printing in the air.
[image]
[ Bottom_3 Generic PETG lw 0.8mm lh 0.25mm if 15.0% ext1 235.0C bed 75.0C.3mf.zip](https://github.com/Ultimaker/Cura/files/11539941/Bottom_3.Generic.PETG.lw.0.8mm.lh.0.25mm.if.15.0.ext1.235.0C.bed.75.0C.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

There's nothing to add here

GregValiant, 2023-05-23:
Thanks for the report and the file.
I can duplicate this in 5.3.1.  It appears that it worked correctly in 4.13.1 (that was the only other version I checked).
@MariMakes of the Cura Team may want to look at this one.

dymbox, 2023-05-26:
> Thanks for the report and the file.
> I can duplicate this in 5.3.1. It appears that it worked correctly in 4.13.1 (that was the only other version I checked).
> @MariMakes of the Cura Team may want to look at this one.

dymbox, 2023-05-26:
> @MariMakes of the Cura Team may want to look at this one

I'm not sure how this works here. Should I contact her directly, or the very mentioning by @MariMakes is enough to draw her in?

GregValiant, 2023-05-26:
That will work.  I think they are busy finalizing the release of 5.4.

MariMakes, 2023-06-05:
👋 MariMakes here!

There is a setting that really helps here called Skin Edge Support Layers. If you have sparse infill it will help with building up extra infill below the skin to support these kind of shapes. In your case it will look like this (if you bring the skin expand distance to 5.0):
[image]

Prusa Slicer released a neat feature called Extend Sparse Infill in their 2.6 beta which extends lines of the bridge infill so that their ends are supported by the sparse infill on the layer below. I can ask the team if that is something we can implement aswell. 

If the forces on the model are limited you can also consider using lightning infill. That will look like this:
[image]

I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

dymbox, 2023-06-06:
Thank you for advice with Skin Edge Support Layers setting. Nevertheless, for general user it's more likely to expect slicer "sane&healthy" decisions, rather than dig for (overcomplicated) solutions and not so clear (named) settings, requiring experimenting & extensive input values tuning before acceptable results show up. As for the Lightning infill, I've tried it but found absolutely unacceptable for model strength against mechanical load. 
Yes, I believe the team should implement similar to Prusa slicer solution. The model under discussion poses no mechanical / structural exotics, so there're great chances for rather often slicing failures due to same issues. To be fair, I was really freaking surprised to discover kinda a slicing error. Thank you for your involvement between users claims & the team! 🙏

ps
you are right, I have Prusa slicer beta installed on my mac. prefer Cura but as we've just seen😏, sometimes need another slicer to get the print job done

pkuiper-ultimaker, 2023-06-21:
Hi dymbox,

Skin edge support must be turned on in the printer profile (Ultimaker printer for example have this on by default). The best route will be to ask your printer supplier (FlyingBear) or somebody from the community to turn this on in the profile of the printer. It will then be on 'automagically' in the next Cura release. 
In the meantime we will also check if we can improve this feature a bit further per your suggestion. Thanks for reporting!

freakydude, 2024-01-26:
@MariMakes I had the same problem with another model I wanted to print. Unfortunately, I saw it on the object after several hours of printing. 

Yes, please do it the Prusa way. You bridge one layer below the top layers exactly between their infill lines where holes in the infill would be in the layer above. All patterns are supported.

CuraSlicer from bottom right. Marked points are start printing completely in air. 
[image]

Same object in PrusaSlicer from bottom right (blue is the bridge, purple is the lowest top layer)

[image]

Holes are closed - yes, in this example the whole object is mostly bridged - but this is special for this object.

Thank you!