Ultimaker/Cura #15594: Windows 11,  Cura  5.3.1 Starts to a white screen...

issacsim, 2023-05-23:
### Application Version

5.3.1

### Platform

Windows 11

### Printer

Qidi XPlus

### Reproduction steps

New member
    0
    4 posts

[Posted May 17](https://community.ultimaker.com/topic/43571-windows-11-cura-531-starts-to-a-white-screen/?do=findComment&comment=322461) (edited) · Windows 11, Cura 5.3.1 Starts to a white screen...

I am getting white screen and no UI, it's not the dual screen issue and I am on the latest nvidia drivers for my video card.

Attached couple of log files, hopefully someone can help.

 

One thing to note, Chrome starts doing the same thing if I try to load Cura first. Without trying Cura first, Chrome works fine until I load Cura

### Actual results

White screen

### Expected results

Normal Cura UI

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/11548612/cura.log)

[stderr.log](https://github.com/Ultimaker/Cura/files/11548615/stderr.log)

MariMakes, 2023-05-26:
Hey @issacsim,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, to hear you are stuck 😞 

I have a couple of questions to narrow it down.
	1. Do you have multiple monitors? Does Cura start if you disable one of the monitors?
	2. Do you have two monitors placed above each other? Can you try this workaround from the community? https://community.ultimaker.com/topic/42633-cura-522-stopped-working/#comment-318105
	3. Do you have GPU Tweak OSD overlay enabled? Does Cura start if you disable that software?
	4. Do you have a Ryzen Graphics card? Can you try this workaround? https://support.makerbot.com/s/article/1667412437627
	5. Do you have the Nahimic audio enhancement service enabled? Can you try this workaround? https://github.com/Ultimaker/Cura/issues/12102#issuecomment-1157907665

issacsim, 2023-05-28:
Thank you for your help,

1, I do have dual screens, tried disabling, unplugging and rebooting one a the time, no luck. 
2. No, they are side by side, tried the workaround, did not help.
3. No GPU Tweaks.
4. Using Nvidia GeForce GTX 980 Ti, tried one monitor only on the onboard Intel GPU, same issue.
5. I do not have Nahimic  service installed on my PC.

Thank You!

MariMakes, 2023-06-05:
Hey @issacsim,

Thanks for the update. This often resolves the issue. 
We have on more thing we can try. 
It's a bit hard for us to test, but we have a nightly build available.

Can you please try it out and let us know if it resolves your issue?
You can find the download links and the discussion here: https://github.com/Ultimaker/Cura/discussions/15098

issacsim, 2023-06-05:
Thank you @MariMakes 

I am still getting the same problem with 5.40.
Every time I try to ruing Cura and get the white screen, my PC becomes unstable and needs to be hard reset. 
This does not happen with anything else I have installed on my PC, only Cura.