Ultimaker/Cura #15597: A model touch the base even it is located at higher position.

junpeinaito, 2023-05-24:
### Application Version

5.3.1

### Platform

Windows10

### Printer

Kobra Max

### Reproduction steps

1. The new STL data is opened.
2. Set the data 10mm higher from the base surface
3. Set the slice parameter
4. Slice

### Actual results

The data which was placed 10mm higher touch the surface

### Expected results

The data is located10mm higher than the base surface and create supports under the model

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I asked our colleague to do the same process by using Cura 5.2.x, it didn't happen, so it may happen later version only. 
[AKMAX_0524-Floating bracketB-P2.zip](https://github.com/Ultimaker/Cura/files/11552123/AKMAX_0524-Floating.bracketB-P2.zip)

GregValiant, 2023-05-24:
Thanks for the report and the file.
This is actually a hidden problem that acts like a bug.
- Turn off "Generate Support"
- In the Mesh Fixes section turn off "Remove Empty First Layers"
- Turn "Generate Support" back on.
So the problem is that "Remove Empty FIrst Layers" gets hidden when "Generate Support" is enabled BUT the setting is being used so it should be visible.  There is also no mention in the tooltip balloons that the settings affect each other.
I'll mark this as a duplicate of #14989 and close this as the Cura team has fixed this for 5.4.  I hope you understand.

When you perform the above steps you get this>
[image]

closed 2023-05-24 by GregValiant

junpeinaito, 2023-06-10:
Thank you so much for your clear explanation. 
We could solve the problem