Ultimaker/Cura #15598: Skirt, Brimm error when you scale the model

Bartovak, 2023-05-24:
### Application Version

5.3.1

### Platform

Windows 10

### Printer

Other

### Reproduction steps

Scale model 200 %

### Actual results

Incorect skirt and brim generation.
[image]

### Expected results

Correct skirt generation.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No more info.

GregValiant, 2023-05-24:
Thanks for the report.
It's hard to tell without a project file but I'd say that the bottoms of the models aren't flat.  If you spin the Cura workspace so you can see from the bottom, the "contact patch" of the model is colored cyan.

This is a model that is resting on the build plate.  The skirt/brim of this model would be correct.
[image]

Here I've tipped the model a bit.  You can see that the contact patch doesn't cover the bottom.
[image]

Here is that tipped model sliced.  You can see that the skirt looks to be incorrect, but it is going around the contact patch as it should.
[image]

So this is a model or model placement issue rather than a bug.  If you would attach a zipped project file it would tell more.
I'm going to remove the bug label from this.  I hope you understand.  I'll leave the report open so you can respond.

Bartovak, 2023-05-24:
Hallo,
100% flat

Fab 365 Razor crest files

---------- Původní e-mail ----------

Komu: Ultimaker/Cura ***@***.***>
Kopie: Bartovak ***@***.***>, Author ***@***.***>
Datum: 24. 5. 2023 12:02:31
(Issue #15598)
"

(https://user-images.githubusercontent.com/64202104/240565607-d401ab10-cc7d-4b74-8658-3869e83d0035.png)

(https://user-images.githubusercontent.com/64202104/240566247-051ced60-820f-4ba8-a484-7274f381b51c.png)

(https://user-images.githubusercontent.com/64202104/240566828-27045915-c774-448f-93fd-07ce82c83546.png)

(https://github.com/Ultimaker/Cura/issues/15598#issuecomment-1560822257), or
unsubscribe
"

Bartovak, 2023-05-24:
---------- Původní e-mail ----------

(https://user-images.githubusercontent.com/64202104/240565607-d401ab10-cc7d-4b74-8658-3869e83d0035.png)

(https://user-images.githubusercontent.com/64202104/240566247-051ced60-820f-4ba8-a484-7274f381b51c.png)

(https://user-images.githubusercontent.com/64202104/240566828-27045915-c774-448f-93fd-07ce82c83546.png)

GregValiant, 2023-05-24:
Please post a project file.  What I see in the image is a model that isn't flush on the build plate.

Bartovak, 2023-05-25:
I have alrady send it yesterday by mail.
---------- Původní e-mail ----------

Datum: 25. 5. 2023 1:29:25
(Issue #15598)
"

GregValiant, 2023-05-25:
I'm sorry but that doesn't work here.  You need to post it here on Github.

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant