Ultimaker/Cura #15601: Pressing "Escape key" while in "discard or keep changes" window will save current changes. I don't think this should be a thing.

Genox666, 2023-05-24:
### Application Version

5.3.0

### Platform

Windows 10

### Printer

Creality Ender-5 Plus

### Reproduction steps

1. Open "Print settings" popup menu at the right.
2. Change any settings you want to configure.
3. Click the blue "Compare and save." icon at the top of the menu.
4. Once you are done looking at the changes you made, press the "Escape key" to close the window.

### Actual results

The window is closed and the changes have been saved as if you had clicked on the "Save" button.

### Expected results

The window closes and the changes are not saved. Same as manually clicking on the red X in the upper corner of the window.

Image1: Print settings window with a modification to the profile
[image]

Image2: Discard or keep changes window before pressing escape
[image]

Image3: Print settings window after pressing escape
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

There's really not much more to it than previously described.

Genox666, 2023-06-05:
Thank you for your reply!
Not a major problem, simply a required adaptation. 
Fingers crossed 🤞 indeed.

Bzzz, 2023-11-21:
This f*cked me over today big time with a dozen of experimental settings on an otherwise fine profile (and of course no backup) - please fix...

andrewroth, 2024-09-03:
+1. This has thrown me off as well.

Genox666, 2024-09-03:
> +1. This has thrown me off as well.

I strongly suggest you switch to Orca Slicer, this is now the go to in terms of slicer software for pretty much any and most machines. You won't have as many issues with it.