Ultimaker/Cura #15603: Step file support

onolox, 2023-05-24:
### Is your feature request related to a problem?

Cura needs to open .step files. This type of file are increasingly more used.

### Describe the solution you'd like

Add the .step file type to the application.

### Describe alternatives you've considered

The alternative is to convert step files to stl, not optinal.

### Affected users and/or printers

Everyone.

T3rm1, 2024-03-30:
The alternative is certainly not converting it to stl. The alternative is 3mf. What features are you lacking in 3mf?

T9Air, 2024-05-03:
["3MF files are known for their ability to show a 3D model’s color and texture, but, as with STL files, 3MF files save a model’s body as a triangular mesh, which can’t capture the same level of detail as STEP files."](https://all3dp.com/2/step-file-format-simply-explained/#i-6-step-vs-3mf)

T3rm1, 2024-05-03:
> ["3MF files are known for their ability to show a 3D model’s color and texture, but, as with STL files, 3MF files save a model’s body as a triangular mesh, which can’t capture the same level of detail as STEP files."](https://all3dp.com/2/step-file-format-simply-explained/#i-6-step-vs-3mf)

On the same page: "While the divided geometric mesh in which STL files describe models isn’t perfectly accurate, it’s easily translated by 3D printers. 3D printers and similar machines are excellent at printing regular geometries like the ones found in STL files’ meshes."

and

"STEP files are better for collaborative CAD projects. STL files are more storage-efficient than STEP files but lack the customization and post-download editing that STEP files offer."

As you can see, there is no advantage for a step file in a slicer software.

T9Air, 2024-05-03:
> > ["3MF files are known for their ability to show a 3D model’s color and texture, but, as with STL files, 3MF files save a model’s body as a triangular mesh, which can’t capture the same level of detail as STEP files."](https://all3dp.com/2/step-file-format-simply-explained/#i-6-step-vs-3mf)
> 
> On the same page: "While the divided geometric mesh in which STL files describe models isn’t perfectly accurate, it’s easily translated by 3D printers. 3D printers and similar machines are excellent at printing regular geometries like the ones found in STL files’ meshes."
> 
> and
> 
> "STEP files are better for collaborative CAD projects. STL files are more storage-efficient than STEP files but lack the customization and post-download editing that STEP files offer."
> 
> As you can see, there is no advantage for a step file in a slicer software.
> 
> And one more for comparison to 3MF: "The 3MF file was developed for 3D Systems (the company) and is great for 3D printing, perhaps more so than STEP files. 3MF files take up less space and can contain slicer settings, such as layer height, inside the raw file format"

For very detailed prints, with very small layers, such as using a .025 layer height with a .1 mm nozzle, the step format may me more accurate, depending on the detail and complexity of the print.

GregValiant, 2024-05-04:
You can get resolution like that in a 3mf or STL.  It's just a question of the resolution settings in the export utility.
There have been scanned objects converted to STL and the resolution is so fine that it's ridiculous.  You get something like a 75mm sphere with 3 million triangles and there is no way that the printer can be that precise.  That would be nearly 170 triangles per mm².  And yes ... those sorts of models have shown up here in complaints about slow slicing.
You would need to get down to a .006 nozzle and layer height to try and print it exactly.

The Cura "Mesh Fixes / Resolution" settings are a filter to keep really short line segments out of the gcode.  Those short lines can bloat files and cause herky-jerky motion in the print head.  The surface of that example sphere could require 85 moves to go 1mm.

There is a point in model resolution and it's like a cliff for each individual printer (not individual families of printers but each single printer).  There is a point where the resolution just can't be handled with molten plastic.  That's true whether the model would be imported as an STL, 3mf, or STEP file.