Ultimaker/Cura #15605: file slising problem

elephonk, 2023-05-25:
### Application Version

5.3.1

### Platform

windows 10

### Printer

Ender 3s1

### Reproduction steps

Hi! 
[ezyzip.zip](https://github.com/Ultimaker/Cura/files/11562364/ezyzip.zip)
all models with the name ..screw bottle cap.. are not slicing,  I increase the scale of the model even by 1 percent, then the problem disappears

### Actual results

all models with the name ..screw bottle cap.. are not slicing

### Expected results

I want everything to be like before and everything works

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
https://www.thingiverse.com/thing:5541914/files

GregValiant, 2023-05-25:
Thanks for the report and the file.
I can't duplicate the problem as you project file sliced fine for me.
Instead of adjusting the scale try adjusting the Initial Layer Height.  Small changes like from .20 to .21 can sometimes fool this bug.
I'll mark this as a duplicate and the Cura team will take a look.
[image]

elephonk, 2023-05-25:
Good evening, Greg! Thank you very much, I have succeeded, I will use your
recommendations. Thanks again for the prompt communication and help in
solving the problem.