Ultimaker/Cura #15606: wire printing

ninjamnt, 2023-05-25:
### Cura Version

5.3.1

### Operating System

win 11

### Printer

anycubic kobra neo

### Name any abnormal settings

wire printing

### Describe the location of your model on the buildplate

in the middle of the plate

### Describe your model

something like a bottle which was a test for wireprinting

### Add your .zip here ⬇️

[bottle_-_bottle_repaired.zip](https://github.com/Ultimaker/Cura/files/11567187/bottle_-_bottle_repaired.zip)

GregValiant, 2023-05-25:
Thanks for the report.
I think you will need to use a previous version of Cura.  Wire Printing appears to be broken in 5.3.1.  A rumor is that it will be removed from Cura.
I'll leave the bug label on this.  Maybe someone from the Cura team has a take on it.

GregValiant, 2024-11-20:
"Wire Printing" has been removed from current versions.  I'll close this.

closed 2024-11-20 by GregValiant