Ultimaker/Cura #15607: Slicing Failed

NicNacPattieMax, 2023-05-25:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Creality CR-10Smart

### Describe your model

part of a large helmet

### Add your .zip here ⬇️

[CCR10S_side-4.zip](https://github.com/Ultimaker/Cura/files/11568984/CCR10S_side-4.zip)

GregValiant, 2023-05-25:
Thanks for the report.
It took a while but the slice completed for me.  There has been a lot of that lately.  I didn't find any errors in the model.
The Cura team will take a look.
BTW...This is a 106 hour print so you are in first place for the longest print award.
[image]

NicNacPattieMax, 2023-05-25:
I was eventually able to get it sliced. That file was 1 of 6 files/parts assembled to make the Dark Helmet from the movie Spaceballls.It randomly fails with large prints. I close the application and retry until it eventually slices without fail.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic