Ultimaker/Cura #15609: slicing failed

khalid135, 2023-05-26:
### Cura Version

5.2.2

### Operating System

windows 10 64bit

### Printer

creality cr10

### Name any abnormal settings

attached screenshot

### Describe the location of your model on the buildplate

center

### Describe your model

yes when all parts at center after I make some space between parts for tree support to touch bed I have slice error. 

### Add your .zip here ⬇️

[CCR10_Fan_V4.zip](https://github.com/Ultimaker/Cura/files/11571560/CCR10_Fan_V4.zip)

yes when all parts at center after I make some space between parts for tree support to touch bed I have slice error.

GregValiant, 2023-05-27:
Thanks for the report and the file.
This sliced for me on the first try so I'm unable to duplicate the behavior.
There have been several reports regarding slicing errors where the model (or group in this case) is at the midpoint of the build plate.  You can try moving the group, rotating it, or maybe changing the Initial Layer Height to get it to slice.
I'll mark this as a duplicate for the Cura Team.

khalid135, 2023-05-30:
Thank you for your reply.
I already rearrange it an it is slice without issue.
Have a nice day.

GregValiant, 2023-05-31:
I'll go ahead and close this then.  The Cura team has a lot of examples of this behavior and the workarounds are getting well known.  I hope you understand.

closed 2023-05-31 by GregValiant