Ultimaker/Cura #15610: Model with holes wont get sliced

mfatihy70, 2023-05-26:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

Ultimaker 3

### Name abnormal settings

All default

### Describe model location

It wont fit otherwise

### Describe your model

A rectanguler shape with square holes in it (without holes it slices just fine)

### Add your .zip here ⬇️

[deckel.zip](https://github.com/Ultimaker/Cura/files/11574338/deckel.zip)

GregValiant, 2023-05-26:
Thanks for the report and the file.
I don't see any errors in the model.  5.3.1 hasn't given me an error, but it's been an hour and it doesn't seem to want to finish.
I tried slicing in 4.13.1 and it took a while but did finish.  5.3.1 has issues.
The Cura team will take a look.

mfatihy70, 2023-06-02:
Hi, got an update: Today the file just got sliced (not the same file as before but similar also with holes) eventhough it took a little more than normal (about 30 seconds) Wish u guys good luck on improving and already great software! From: GregValiantSent: Freitag, 26. Mai 2023 14:04To: Ultimaker/CuraCc: M. Fatih Yıldız; AuthorSubject: Re: [Ultimaker/Cura] Model with holes wont get sliced (Issue #15610) Thanks for the report and the file.I don't see any errors in the model. 5.3.1 hasn't given me an error, but it's been an hour and it doesn't seem to want to finish.I tried slicing in 4.13.1 and it took a while but did finish. 5.3.1 has issues.The Cura team will take a look.