Ultimaker/Cura #15612: Skirt generated outside of the print range

konvoj, 2023-05-26:
### Application Version

5.3.0

### Operating System

Windows 10

### Printer

Ultimaker S5, Ultimaker 2+C 

### Reproduction steps

1. Placed objects on the build plate
2. Selected "skirt"

### Actual results

The skirt is generated outside of the print area. This is visible in the UI, but I didn't noticed that.
What happens is that the skirt is printed as multiple line segments with filament being retracted for travel moves in between. 
This completely negates the continuous skirt being pressure-equalizing tool before the start of the first layer. 

### Expected results

Generate continuous skirt. If skirt goes outside of the printable range, notify user. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Btw. "brim" setting behaves similar as skirt, but in brim's case I consider it being a logical wanted behavior.

This must have been a recent change, since in older versions brim and skirt margins were implemented by adding an offset to the part's projection onto a buildplate, which wasn't technically a clean solution. But making the buildplate adhesion functionally an offset of an object, this resulted in an error message, that the object is outside of the printable volume. 

Please don't go back to that hacky solution, the new algorithm works better, just needs a proper boundary check.

The new implementation probably also causes my previous issue https://github.com/Ultimaker/Cura/issues/15552, since adhesion features are now not being considered just an offset of the object itself.

[image]
[image]
[image]
[Bug_Print Files.zip](https://github.com/Ultimaker/Cura/files/11574645/Bug_Print.Files.zip)

konvoj, 2023-06-01:
Agree in case of brim, for skirt, this shall be a bug IMHO.

MariMakes, 2023-06-05:
Hey @konvoj,

Thanks for your report 👍

This behavior changed indeed in Cura 5.3 from being blocked to slice to having the adhesion be cut of for you. You can read more about it here: https://github.com/Ultimaker/CuraEngine/pull/1613.

I can ask the team if we can include a warning if your skirt is not a continuous line or is missing too much. 
I'll bring it up to the team to see what they can do to improve it. 
Fingers crossed 🤞

GregValiant, 2023-06-06:
It seems to allow a user to utilize the entire build surface rather than having the Build Plate Adhesion take up room.  The purge lines in my start Gcode are enough to get the flow going so I often set the build plate adhesion to "None".  When I use a skirt it makes little difference whether I throw away 2 small pieces or 1 large piece.  It's scrap either way.

konvoj, 2023-06-06:
The statement on the disconnected skirt lines being a non-issue is assuming the use of direct drive printer. Note that on Ultimakers and other bowden setups, the material flow needs to be stabilized by printing a continuous line with as close the extrusion rate as the future lines of the first layer to equalize the pressure in the nozzle. Bringing multiple retracts and inserts into this completely undermines the flow equalization.

You are correct that when user prints really big, then she must make sure that buildplate adhesion is set to "None" anyway or carefully program-in a purge line; This statement is not contradicting the fact that when buildplate adhesion is set to "Skirt", the user shall be aware  of the fact that she can't utilize the buildplate area completely. The skirt from its definition shall print continuous line(s) imho. Potential improvement point is to inform users that don't understand the relation between using any builplate adhesion choice other than "None", that using skirt or brim will not allow to print objects with dimensions that are close to the dimensions of the buildplate. Yet another improvement would be if Cura can detect that due to the buildplate adhesion setting the object doesn't fit the buildplate and offer to the user choice to choose buildplate adhesion "None". This might come with a potential failure point for BFUs, where they disable buildplate adhesion to print big, while not knowing the buildplate adhesion challenges of printing big parts and thus lead to failed prints.

For further reference, there's an episode of "The Additive Chef" on printing big parts using Ultimaker printers:
https://web-staging.ultimaker.com/learn/printing-big-parts-and-speed-tweaks-webinar