Ultimaker/Cura #15615: Overextrude 2nd layer above supports

Behem07h, 2023-05-27:
### Is your feature request related to a problem?

When using supports, the first layer above the supports inevitably doesnt squish as much as a normal layer, which causes layer adhesion issues between it and the layers directly above it. It also makes the supported surface look sparse and ugly.

### Describe the solution you'd like

I want to specifically increase the flow on the 2nd and possibly following bottom skins above a supported surface. This will allow the layers after the first to better adhere and fill the gaps between the lines of the layer below them. 

### Describe alternatives you've considered

An existing alternative solution would be to simply decrease the z gap between the support and the supported area, however this is limited to a multiple of the layer height unless using adaptive layers, and also causes the support to fuse more strongly to the object which makes support removal difficult

### Affected users and/or printers

FDM printers will benefit from these changes as supported areas should have a cleaner appearance

### Additional information & file uploads

my suggestion would allow supported areas to be sliced as follows:

100% flow, third bottom skin
110% flow, second bottom skin
100% flow, first bottom skin
100% flow, last support layer

perhaps an option to iron the nth layer of bottom skin may also help, however i did not include it in the main body of my suggestion as that is just a guess

GregValiant, 2023-05-27:
- _"...and also causes the support to fuse more strongly to the object"_
-  _"...should have a cleaner appearance"_

Those would appear to be mutually exclusive without a dual extruder printer and dissimilar materials.  They are in fact the reasons people move to dual extruder printers.
100% Support Interface Density and a Z gap of 0.0 will give you a nice looking underside that you will not be able to divorce from the support unless the materials are different enough that they don't bond together.

If you turn on the Bridge Settings there are flow settings that will do what you want, but my first point stands.

Behem07h, 2023-05-27:
"...and also causes the support to fuse more strongly to the object"
The reason why I put the above under "alternatives you've considered" is precisely because its an alternative that doesnt work cleanly without using dual extruders. Is the "alternatives you've considered" section not meant for people to express that they've tried xyz alternative things and they didnt work, to justify their feature request?

If bridge settings do indeed affect skins above supports then that will indeed do what i want, however imo this should be made much more clear in it's description as in other slicers (prusa slicer, super slicer and derivatives) and in most people's minds, bridges and supported areas are two different things.

Behem07h, 2023-05-28:
thinking back, bridge and supported areas should definitely be separated, because overhangs are certainly not bridges but can still be supported areas. Are you very certain that the bridge flow settings will affect *all* of the regions above supports?

GregValiant, 2023-05-28:
I designed this "support testing" model just to play with different settings.
The area shown is cantilevered and so the normal "bridge" settings would exclude it as not worthy of a bridge.
With the bridge "Skin Support Threshold" at 100% it is considered "bridgable" and you can see that there is bridging.  This is with the bridge skin and wall flows at 50%.  (I'm pretty sure that there is no way that those end corners would stay in place.  They would drag behind the nozzle and end up cutting across the corner.)
[image]

Here I have adjusted the bridge skin and wall flows to 150%,
[image]

The only way I can think of to get a "nice" finish on the underside of a supported area, is to get squish to the support roof and the support roof would need to be 100% density.
If there is an air gap of 1 layer and you set the flow to 200% then you will get squish because you have effectively created a double thick layer and taken up the air gap.  It will then glue itself to the support and we're back at square one.

This view is different.  The bridge settings are off but I've brought in a support blocker and configured it as a Cutting Mesh with 200% flow.  The blocker is 1 layer height thick and covers layer 2 of this overhang.
Here is the first layer over the support at 100% flow.
[image]

And here is the second layer over the support that is being affected by the blocker and is at 200% flow.
[image]

Because the area is cantilevered:  Gravity along with the push from the double flow will cause it to end up on the support interface.  My guess would be that it would glue itself there.

Time for another cup of coffee.