Ultimaker/Cura #15616: Print settings/ Travel. In the Z Hop

caramelo72, 2023-05-27:
### Is your feature request related to a problem?

No

### Describe the solution you'd like

Hello, in print settings/ Travel. In the Z Hop Height. Will be nice increment option to stop doing ``Z Hop Height`` at Layer we need to stop the ``Z Hop Height``. 
Because in some works only this option is need in the first layers. 
 And is not nesseary this job in the all printing process. 
Will reduce printing time and 
Thank you. 
Paulo Alexandre

### Describe alternatives you've considered

Increment option to stop doing ``Z Hop Height`` at Layer we need to stop the ``Z Hop Height``. 

### Affected users and/or printers

No

GregValiant, 2023-05-27:
I have proposed this "AlterZHops" post processor.  We'll see if it makes it into Cura.
Unzip and put the .py file into the Scripts folder in your configuration folder.
On my Windows system it would go in "C:\Users\........\AppData\Roaming\cura\...version...\scripts
You can be a tester.  Let me know if you find issues or have a suggestion.

[AlterZhops.zip](https://github.com/Ultimaker/Cura/files/11582189/AlterZhops.zip)

caramelo72, 2023-05-28:
Hello, Thank you for your anser .Go teste and give you the resultAll the best to youPaulo

Enviado do Yahoo Mail para Android

caramelo72, 2023-05-28:
Hello good afternoon.
The default z hop start in the first layer.
I just think option necessary to indicate the end of the z hop
Like ex:
Z HOP STOP AT LAYER    0.06mm
Paulo Alexandre

GregValiant, 2023-05-28:
You would set the post processor to "0" zhop height and then the start layer to whatever layer you want to end the zhops at.  When the End Layer is set to -1 your change would continue to the end of the file.

For example, you want Zhops for the first layer only, and then want them to stop for the rest of the file.  Set the zhop height to 0 and the Start Layer to 2 with the end layer set to -1.

Every situation is different.  I don't want zhops at the beginning of a file, only when large horizontal holes close up and create feather edges that curl up.  I don't want the nozzle hitting those curls.
I might set the Zhop height to 0 for layers 1 to 25 and then use another instance of the post processor and shut them off from 35 to 90, and then a third instance of the post processor and shut them off from layer 115 to the end of the file.

Doing it another way would require altering the Cura Engine or creating a full plugin, both of which are beyond me.

caramelo72, 2023-05-28:
HelloGot it :-)The whay to put the scriptis diferent from yours.go testAnd report the workThank youPaulo

caramelo72, 2023-05-29:
HelloTested in 3 printers.Work like a charm.Nice job.
I have a Artilery x2 can you had this printer in cura.Thank youPaulo

GregValiant, 2023-05-29:
The printer definitions come from a member of the community or the manufacturer.  It's necessary to have one of the printers and the full specs in order to come up with a definition file, extruder files, mesh files, nozzle files, quality files, etc.