Ultimaker/Cura #15617: Slicing Failed

Genius20000, 2023-05-27:
### Cura Version

5.3.1

### Operating System

MacOS Ventura 13.3.1

### Printer

Anycubic Kobra Neo

### Describe model location

automatic center

### Describe your model

First printer first project

### Add your .zip here ⬇️

[CFFFP_Corpo1.3mf.zip](https://github.com/Ultimaker/Cura/files/11582235/CFFFP_Corpo1.3mf.zip)

GregValiant, 2023-05-27:
Thanks for the report and the file.
I didn't have any trouble slicing the project on Windows but there have been several reports lately of slicing failing on MAC systems.
I'm going to mark this as a duplicate slicing error and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic